OctoPrint/OctoPrint-Slic3r #1: Profile parser fails to handle Slic3r config files for machines with multiple extruders

C-o-r-E, 2015-05-12:
Using branch devel and latest octoprint-slic3r master.

For certain parameters in the Slic3r config, values are comma separated. It looks like profile.py does not expect that. Here are some traces from the log:

```
2015-05-12 16:28:11,368 - plugins.slic3r.slic3r.profile - ERROR - Got an exception while trying to convert the value for extru$
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_slic3r/profile.py", line 252, in convert_value
    return float(value)
ValueError: invalid literal for float(): 1,1
2015-05-12 16:28:11,371 - plugins.slic3r.slic3r.profile - ERROR - Got an exception while trying to convert the value for filam$
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_slic3r/profile.py", line 252, in convert_value
    return float(value)
ValueError: invalid literal for float(): 1.75,3
...
...
```

feiming, 2015-05-14:
Can you share your Slic3r profile?

C-o-r-E, 2015-05-14:
I purposely truncated the trace as it exposes the pattern that I think is key.

What I mean by that is... Take this line from my snippet above:

'''
ValueError: invalid literal for float(): 1,1
'''

That tells me somewhere in the parsing of the profile, we get a string and try to convert it to a float without first checking what the string looks like. At the very least we should be checking for commas on that line and others because the profile (or slic3r config) is a valid file generated by slic3r.

You can find a link to the profile (config file) [here](https://gist.github.com/C-o-r-E/d318ffde37f9dacc2cd8).

I should be able to take a look at the code eventually. Just have my hands full for the moment.

feiming, 2015-05-14:
Please test my change.

    pip install https://github.com/feiming/OctoPrint-Slic3r/archive/master.zip

feiming, 2015-05-15:
I tried a few way to handle the conversion. But the merged profile from octoprint is different.some parameters werev overrided by default.

fibasile, 2015-09-09:
Same here, slicer ini is totally different from the one loaded. Can't we just use the uploaded one? I.e. no merging for defaults

foosel, 2015-10-19:
I don't have the resources to look into this at the moment (hence the big fat warning in the README), but if anyone wants to give it a whirl to switch to directly using the regular ini, I'll happily accept pull requests.

The problem here is that in the long run (even though the UI currently doesn't reflect this), the idea is to allow the user to adjust certain slicing parameters (e.g. temperature, filament diameter etc) per individual slicing request. Also, the machine parameters should be taken from the printer profile within OctoPrint instead of having to upload several slicing profiles (in situations where one OctoPrint instance services - one at a time - multiple printers). Hence the conversion into an internal format with known parameter names, to be able to overwrite those. But I guess it would also be perfectly fine to only merge the overwritten parameters with the values from the native ini file.

In fact, my colleague @nicanor-romero is currently looking into doing exactly this for a new version of the CuraEngine integration, since the interface and format there changes so often everything else is just not feasible anymore. Judging by the problems everyone here is reporting, we have a similar situation with Slic3r.

eyal0, 2016-04-20:
**The** problem is that this plugin assumes defaults for each parameter to Slic3r, for example, nozzle_width and bed_temperature.  When there are new versions of Slic3r, the plug in needs to be updated to include the new defaults because any settings that the plugin hasn't heard of is **deleted**.  This can screw up your build.

I wrote a version of the plug in that doesn't assume defaults.  It mostly just stores the configuration that it gets and uses it for slicing.  It doesn't have the above mentioned ability to let users modify profile features but it should work.

https://github.com/eyal0/OctoPrint-Slic3r

eyal0, 2017-02-07:
Please try again.  OctoPrint-Slicer no longer tries to parse the profile, it simply uses the provided profile as is.  Let me know if there are still problems.  Thanks!

eyal0, 2017-04-04:
No response in a long time.  Closing this.

closed 2017-04-04 by eyal0