OctoPrint/OctoPrint-Slic3r #6: Status?

fromagek, 2016-12-20:
What is the current status of this project.
I really would like to use Slic3r on Octoprint. What are the limitations of the current code.
Where must be fixes or improvements. I am willing to contribute, but would like to get an overview before I dig into.
Best Florian

m0n5t3r, 2017-01-03:
I would guess most people came to the conclusion that slic3r's octoprint API integration ("send to printer") is a better solution, and this project is abandoned :) 

(I arrived here while doing a bit of housekeeping, I figured it would be a good idea to pay a visit to the plugin's homepage before uninstalling)

foosel, 2017-01-09:
Personally I don't use slic3r myself, and the issues I ran into when trying to get this release ready (weird error codes returned by the external process) are the reason this never was officially released. By now I also guess that the whole profile implementation would need to be updated to current versions.

I don't see myself having time to actually look into getting this looked into for a long time to come (development and maintenance of OctoPrint itself pretty much eats up my time completely). If anyone wants to adopt this, feel free.

eyal0, 2017-02-06:
Javierma has done some good work and I've built upon that, too.  Slicing with slic3r through octoprint is very good now and Slic3r Prusa Edition is the recommended slicer for Prusa printers.  My fork of OctoPrint-Slic3r works perfectly with octoprint and slic3r.

@foosel , would you consider merging javierma's work or mine?  It's a shame that the OctoPrint-Slic3r plugin in the official repository isn't functional when there are working alternatives.

foosel, 2017-02-07:
@eyal0 merging it is not the problem, ongoing maintenance is what I simply can't find time for, especially considering that I don't actually use Slic3r myself. Would you maybe be open to adopt the plugin? I'd give you write access here to get it into shape and then it could actually be pushed on the official plugin repository (it isn't at the moment).

eyal0, 2017-02-07:
Gladly! I've got a fork that works well. Either I'll merge to yours or put
mine in the repository.

Eyal

foosel, 2017-02-07:
:+1: You should have gotten an invite. If you could also take a good look at the open PR from Javier that would be great (if that isn't already merged into your fork anyhow).

Need to get back on my couch now, stupid cold.

eyal0, 2017-02-07:
Got it!  I'll get to work.  Thanks!

eyal0, 2017-02-07:
Done.  It should work now.  For my testing, I have sliced from the command line and via this plugin and both outputs are identical.  Most of it is based on @Javierma 's work plus a few changes from me to support other versions of slic3r.

If you see any bugs, let me know.  For example, if slicing with the plugin and slicing directly from slic3r make different outputs.  Thanks.

closed 2017-02-07 by eyal0

fromagek, 2017-02-07:
@eyal0 It’s rally nice! Thanks a lot. Will give it a try and report tomorrow.