OctoPrint/OctoPrint-Slic3r #7: Slic3r does not appear in Slicer drop down menu.

fromagek, 2017-02-08:
Slic3r does not appear in Slicer drop down menu.
I installed Slic3r according to the README.md on RPi 3 without issues.
However, when I try to slice a STL I can only choose CuraEngine in the drop down menu.

best florian

eyal0, 2017-02-08:
And you have installed the OctoPrint-Slic3r plugin?  It will only work if slic3r is already configured so you must have the executable path set and also a profile uploaded.

If you have done all that and it still doesn't work, press F12 to open the developer console.  Then reload the page and try again.  Send me a copy of all the log messages and especially any errors.  Thanks.

fromagek, 2017-02-08:
Hey eyal0,
Just found out that it must somehow have to do with the cache of my Firefox.
Using another browser or Firefox in private mote everything looks fine.

closed 2017-02-08 by fromagek

eyal0, 2017-02-08:
Glad to hear it.  Enjoy!