OctoPrint/OctoPrint-Slic3r #9: Validated with Prusa Beta

synman, 2017-03-05:
I apologize as this is far from an issue.  But I can confirm this plugin works wonderfully with the Prusa Beta version of Slic3r.

Here's my variation of your shellscript that pulls in the necessary additional dependencies and uses their fork:

```
pi@octopi:~ $ cat install_slic3r.sh 
#!/bin/sh

echo "Shell script created by Javier Martínez Arrieta for Slic3r installation\n"
#Ask the user for a version
#echo "Please indicate which version you desire to be installed (e.g. 1.1.7 or version_1.33.5 if building prusa)"
#read version
echo "The installation of Slic3r takes a long time. Please be patient"
mkdir $HOME/prusa
cd $HOME/prusa
echo "Installing required libraries and dependencies..."
sudo apt-get install git libboost-all-dev libboost-geometry-utils-perl libboost-system-dev libboost-thread-dev git-core build-essential libgtk2.0-dev libwxgtk2.8-dev libwx-perl libmodule-build-perl libnet-dbus-perl cpanminus libextutils-cbuilder-perl gcc-4.7 g++-4.7 libwx-perl libperl-dev
sudo cpanm AAR/Boost-Geometry-Utils-0.06.tar.gz Math::Clipper Math::ConvexHull Math::ConvexHull::MonotoneChain Math::Geometry::Voronoi Math::PlanePath Moo IO::Scalar Class::XSAccessor Growl::GNTP XML::SAX::ExpatXS PAR::Packer
echo "Cloning Slic3r repository..."
#git clone https://github.com/alexrj/Slic3r.git
git clone https://github.com/prusa3d/Slic3r.git
cd Slic3r
#git checkout $version
echo "Building and testing Scli3r..."
sudo perl Build.PL
echo "If everything was installed properly,you should be able to run Slic3r with the command ./slic3r.pl"
```

And my slic3r.pl output:

```
pi@octopi:~/Slic3r $ ./slic3r.pl --version
1.31.6
```

```
pi@octopi:~/Slic3r $ DISPLAY=:0 ./slic3r.pl 
Slic3r 1.31.6 is a STL-to-GCODE translator for RepRap 3D printers
written by Alessandro Ranellucci <aar@cpan.org> - http://slic3r.org/

Usage: slic3r.pl [ OPTIONS ] [ file.stl ] [ file2.stl ] ...

    --help              Output this usage screen and exit
    --version           Output the version of Slic3r and exit
    --save <file>       Save configuration to the specified file
    --load <file>       Load configuration from the specified file. It can be used 
                        more than once to load options from multiple files.
    -o, --output <file> File to output gcode to (by default, the file will be saved
                        into the same directory as the input file using the
                        --output-filename-format to generate the filename.) If a
                        directory is specified for this option, the output will
                        be saved under that directory, and the filename will be
                        generated by --output-filename-format.
  
  Non-slicing actions (no G-code will be generated):
    --repair            Repair given STL files and save them as <name>_fixed.obj
    --cut <z>           Cut given input files at given Z (relative) and export
                        them as <name>_upper.stl and <name>_lower.stl
    --split             Split the shells contained in given STL file into several STL files
    --info              Output information about the supplied file(s) and exit
    
    -j, --threads <num> Number of threads to use (1+, default: 4)

  GUI options:
    --gui               Forces the GUI launch instead of command line slicing (if you
                        supply a model file, it will be loaded into the plater)
    --no-plater         Disable the plater tab
    --gui-mode          Overrides the configured mode (simple/expert)
    --autosave <file>   Automatically export current configuration to the specified file

[... 249 lines cut ...]
    --support-material-extrusion-width
                        Set a different extrusion width for support material
    --infill-overlap    Overlap between infill and perimeters (default: 55%)
    --bridge-flow-ratio Multiplier for extrusion when bridging (> 0, default: 1)
  
   Multiple extruder options:
    --extruder-offset   Offset of each extruder, if firmware doesn't handle the displacement
                        (can be specified multiple times, default: 0x0)
    --perimeter-extruder
                        Extruder to use for perimeters and brim (1+, default: 1)
    --infill-extruder   Extruder to use for infill (1+, default: 1)
    --solid-infill-extruder   Extruder to use for solid infill (1+, default: 1)
    --support-material-extruder
                        Extruder to use for support material, raft and skirt 
                        (1+, 0 to use the current extruder to minimize tool changes, default: 1)
    --support-material-interface-extruder
                        Extruder to use for support material interface 
                        (1+, 0 to use the current extruder to minimize tool changes, default: 1)
    --ooze-prevention   Drop temperature and park extruders outside a full skirt for automatic wiping
                        (default: no)
    --standby-temperature-delta
                        Temperature difference to be applied when an extruder is not active and
                        --ooze-prevention is enabled (default: -5)

```

eyal0, 2017-03-08:
I also use the Prusa Beta version.  Thanks for the updates to the script.

realbuz, 2017-03-09:
trying to use this on octopi on a raspberry 3 and I get the following: 

Configuring /home/pi/prusa/Slic3r/xs ... Slic3r requires the Intel Thread Building Blocks libraries.                                        Please make sure the library is installed.

If the Intel Thread Building Blocks library is installed, this script should be able to locate them i                                       n several
standard locations. If this is not the case, you might want to supply a path to the library
through the TBB_DIR environment variable:

    TBB_DIR=/path/to/TBB perl Build.PL

Or you may specify TBB_INCLUDEPATH and TBB_LIBRARYPATH separatly, which
is handy, if you have built the Thread Building Blocks libraries with mutliple settings.

N/A
! Configure failed for ./xs. See /root/.cpanm/work/1489031000.30517/build.log for details.
The XS/C++ code failed to compile, aborting

is this not compatible ?

realbuz, 2017-03-09:
I had to follow this to go forward:

https://www.theimpossiblecode.com/blog/intel-tbb-on-raspberry-pi/

realbuz, 2017-03-09:
spoke too soon, got much further.. but results: 

gleMesh.o src/libslic3r/TriangleMesh.cpp
In file included from /usr/local/include/tbb/tbb_machine.h:247:0,
                 from /usr/local/include/tbb/task.h:25,
                 from /usr/local/include/tbb/parallel_for.h:25,
                 from src/libslic3r/TriangleMesh.cpp:16:
/usr/local/include/tbb/machine/gcc_armv7.h:31:2: error: #error compilation requires an ARMv7-a architecture.
 #error compilation requires an ARMv7-a architecture.
  ^
error building src/libslic3r/TriangleMesh.o from 'src/libslic3r/TriangleMesh.cpp' at /usr/share/perl5/ExtUtils/CBuilder/Base.pm line 172.
FAIL
! Installing ./xs failed. See /root/.cpanm/work/1489036678.3101/build.log for details. Retry with --force to force install it.

eyal0, 2017-03-14:
I compiled on an older RPi successfully.  Nothing about requiring ARM7, though.  I used OctoPi build.

synman, 2017-03-14:
My build was successful (I've actually built a couple different Prusa releases now) using the attached script on my Pi 3.

eyal0, 2017-04-04:
@realbuz TBB is a new requirement for Pursa3d Slic3r version 1.34.0.  Trying building version_1.33.8 and see if that works.

eyal0, 2017-05-01:
@realbuz and @synman I put up new instructions.  Just copy and paste: https://github.com/OctoPrint/OctoPrint-Slic3r/wiki/How-to-install-Slic3r-on-RPi

Let me know how it works if you try and go ahead and make changes if you need.

closed 2017-05-01 by eyal0