OctoPrint/OctoPrint-Slic3r #10: Installation of the plugins fail

fidel3d, 2017-03-07:
Even though the installation of the plugin goes smooth I get an error popup. I will attach the text from the installer and the error.

Please advise,

Thanks
[out_from_installer.txt](https://github.com/OctoPrint/OctoPrint-Slic3r/files/824816/out_from_installer.txt)
[image]

fidel3d, 2017-03-07:
I forgot to add the following:
OctoPrint version 1.3.1
and I chose the slic3r ver 1.1.7

foosel, 2017-03-07:
You already have the plugin installed:

```
Requirement already satisfied (use --upgrade to upgrade): OctoPrint-Slic3r==1.0.0 from https://github.com/OctoPrint/OctoPrint-Slic3r/archive/master.zip in /home/pi/oprint/lib/python2.7/site-packages
```

Try uninstalling it first, then install manually or even better, use the Plugin Repository (Settings > Plugin Manager > Get More...) instead of installing manually.

spmp, 2017-03-27:
Maybe a different issue, but I cannot install either:
```
Installing plugin from uploaded archive...
/usr/bin/python -m pip install /tmp/tmp8hMKjw.zip --user
Processing /tmp/tmp8hMKjw.zip
Collecting OctoPrint (from OctoPrint-Slic3r===1.0.0)
Could not find a version that satisfies the requirement OctoPrint (from OctoPrint-Slic3r===1.0.0) (from versions: )
No matching distribution found for OctoPrint (from OctoPrint-Slic3r===1.0.0)
You are using pip version 8.1.2, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
Done!
```

eyal0, 2017-04-04:
Are you doing it on an RPi using OctoPi?  I usually run pip from inside the virtual environment like this:

`./venv/bin/pip`

Can you check if your OctoPrint is actually in a virtual environment?  Try running:

`ps aux | grep octo`

And see if `venv` is in the path.

eyal0, 2017-05-27:
No response in a long time.  Maybe the problem went away?  Re-open if there is still an issue.

closed 2017-05-27 by eyal0

nitrag, 2018-01-27:
For me I had to use `sudo`, just like I installed (non-venv) via sudo