OctoPrint/OctoPrint-Slic3r #11: More configuration options please!

nathanielstenzel, 2017-04-04:
The program can take a large amount of configuration options on the command line. Please implement some of them on the basic and advanced configuration tabs for the slicer.

eyal0, 2017-04-04:
This slicer, like the cura slicer, works by using stored profiles which you can add in the configuration.  If you want to have multiple configurations, you should upload multiple profiles.

There are no basic and advanced tabs in this project.  Maybe you are thinking of OctoPrint-Slicer (with an "e", not a "3")?

dermeb, 2017-04-05:
It would be nice, if you could switch between the different (imported) print , filament and printer settings, like in Slic3r. Maybe I oversee something? I do not want to modify them, but choose my imported settings.

eyal0, 2017-04-05:
When you try to slice, a dialog opens where you can select a "Slicing Profile".  It's in this image: https://camo.githubusercontent.com/9fc508ca40c09d6a885a5803c43402a7d69ae9fe/687474703a2f2f692e696d6775722e636f6d2f4143316730756e2e706e67

If you upload a few slicing profiles, you can use have one for each filament and print settings combinations.

Is that not enough?

foosel, 2017-04-05:
> Is that not enough?

And if the answer to that is "yes", try [this additional plugin](http://plugins.octoprint.org/plugins/slicer/). Still unhappy? I'm sure PRs are welcome =)

dermeb, 2017-04-05:
> If you upload a few slicing profiles, you can use have one for each filament and print settings combinations. 

Ah now I see. If you export config via Slic3r you only get the combination from the current print , filament and printer settings. I thought that you would export all available settings (which can be done as Config bundle), so the step to choose the ones was mssing to me. Now I understand ;)

> try this additional plugin. Still unhappy? I'm sure PRs are welcome =)

Yes through that plugin I was made aware of the slic3r-plugin ;)
Unfortunately I'm a bad programmer, but I'm trying to support the community on other fronts ;)
BTW, thanks for the awesome work of octoprint, I really love to use it!

nathanielstenzel, 2017-04-07:
Yeah. The interface that I was looking at was the generic slicer plugin. I imagine this plugin would not work without that one right? I thought maybe this plugin told the other plugin what to show and took slicing commands from it. Am I misunderstanding things?

eyal0, 2017-04-14:
There are two plugins:
1. This one, _Slic3r_, lets you use slic3r for slicing models.
2. The other one, _Slicer_, lets you have a graphical interface.

Slicer users the slic3r plugin, and also the built-in cura plugin, to do slicing.  So if you want GUI and you want slic3r, you need both.

The slicer GUI also has options for modifying a slicing profile slightly. For example, you can change the speed, flow rate, etc.  Just the most common parameters can be changed.

I hope that clears it up.  If there are no further questions, can this issue be closed?

nathanielstenzel, 2017-04-14:
If the slic3r plugin was to add parameters that are not in the profile, would the slicer plugin then let you modify them so that the slic3r plugin could use them? If that does not sound plausible, please feel free to close it. If that is plausible, I think it would be cool to try it out.

eyal0, 2017-04-14:
No, it wouldn't work. The slicer plugin takes over from the slic3r plugin,
taking only the stored profiles but no parameter changes.

Could you work up a mock-up of how you'd like it to work? Because, so far,
it seems like what you need is to install both plugins and that it. It's a
pretty good combination, using both of them.

eyal0, 2017-06-11:
No response in a long time.

closed 2017-06-11 by eyal0