OctoPrint/OctoPrint-Slic3r #13: Cannot complete install on Raspberry Pi Zero with WiFi

impartit, 2017-04-13:
After 12 hours or more the script errors with 
cc -I/usr/lib/arm-linux-gnueabihf/perl/5.20/CORE -DVERSION="0.01" -DXS_VERSION="0.01" -fPIC -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fwrapv -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -xc++ -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -c -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fwrapv -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g -o buildtmp/XS.o buildtmp/XS.c
The XS/C++ code failed to compile, aborting
error building buildtmp/XS.o from 'buildtmp/XS.c' at /usr/share/perl/5.20/ExtUtils/CBuilder/Base.pm line 175.

This after a long protracted attempt at installing Slic3r with the Slicer plugin on OctoPrint. Pi OS is the latest OctoPrint version 1.3.2, 0.14.

Please, please advise

eyal0, 2017-04-14:
Yes, it takes a long time.  Which version did you try to install?

impartit, 2017-04-14:
Which version of what?

eyal0, 2017-04-14:
Which version of slic3r did you try to install?

Did you run `git clone` to check it out?  What did you clone?

Can you provide the rest of the log, too?  Did you already try all the `sudo apt-get` as recommended in the instructions?

NiklasOlofsson, 2017-04-15:
I looked at your script and installed all the dependencies, running on Pi 2.
Used the Prusa Edition by using this commands:

```
git clone https://github.com/prusa3d/Slic3r.git
cd Slic3r/
git checkout version_1.34.1

sudo apt-get install git libboost-all-dev libboost-geometry-utils-perl libboost-system-dev libboost-thread-dev git-core build-essential libgtk2.0-dev libwxgtk2.8-dev libwx-perl libmodule-build-perl libnet-dbus-perl cpanminus libextutils-cbuilder-perl gcc-4.7 g++-4.7 libwx-perl libperl-dev

sudo cpanm AAR/Boost-Geometry-Utils-0.06.tar.gz Math::Clipper Math::ConvexHull Math::ConvexHull::MonotoneChain Math::Geometry::Voronoi Math::PlanePath Moo IO::Scalar Class::XSAccessor Growl::GNTP XML::SAX::ExpatXS PAR::Packer
```

Output from sudo perl Build.PL:
```
You have make /usr/bin/make
You have LWP 6.08
You have /bin/tar: tar (GNU tar) 1.27.1
Copyright © 2013 Free Software Foundation, Inc.
Licens GPLv3+: GNU GPL version 3 eller senare <http://gnu.org/licenses/gpl.html>.
Detta program är fri programvara.  Du kan modifiera och distribuera den.
Det finns inte NÅGON SOM HELST GARANTI, till den grad som lagen tillåter.
Skrivet av John Gilmore och Jay Fenlason.
You have /usr/bin/unzip
--> Working on ./xs
Entering /home/pi/Slic3r/xs
Checking if you have Module::Build 0.36 ... Yes (0.4210)
Running Build.PL
Configuring /home/pi/Slic3r/xs ... Slic3r will be built with GUI support
Possible precedence issue with control flow operator at /usr/lib/arm-linux-gnueabihf/perl5/5.20/Alien/wxWidgets/Utility.pm line 77.
Use of uninitialized value in concatenation (.) or string at /usr/local/share/perl/5.20.2/Devel/CheckLib.pm line 411.
Use of uninitialized value in concatenation (.) or string at /usr/local/share/perl/5.20.2/Devel/CheckLib.pm line 411.
Use of uninitialized value in concatenation (.) or string at /usr/local/share/perl/5.20.2/Devel/CheckLib.pm line 411.
Use of uninitialized value in concatenation (.) or string at /usr/local/share/perl/5.20.2/Devel/CheckLib.pm line 411.
Use of uninitialized value in concatenation (.) or string at /usr/local/share/perl/5.20.2/Devel/CheckLib.pm line 411.
Slic3r requires the Intel Thread Building Blocks libraries. Please make sure the library is installed.

If the Intel Thread Building Blocks library is installed, this script should be able to locate them in several
standard locations. If this is not the case, you might want to supply a path to the library
through the TBB_DIR environment variable:

    TBB_DIR=/path/to/TBB perl Build.PL

Or you may specify TBB_INCLUDEDIR and TBB_LIBRARYDIR separatly, which
is handy, if you have built the Thread Building Blocks libraries with mutliple settings.

N/A
! Configure failed for ./xs. See /root/.cpanm/work/1492286248.4710/build.log for details.
The XS/C++ code failed to compile, aborting
```

eyal0, 2017-04-16:
Yes, versions >= 1.34 don't work without a fix for the TBB. I'm working on it.  version 1.33.8 will work.

NiklasOlofsson, 2017-04-16:
Great, thanks! Looking forward to it working.
Btw, the checkout needs "version_" added before the number (like git checkout version_1.34.1) if using the Prusa edition.

NiklasOlofsson, 2017-04-17:
Found this:
https://www.theimpossiblecode.com/blog/intel-tbb-on-raspberry-pi/
followed the instructions (precompiled deb will probably work for Pi 2 and 3 at least).
Also, I did this change manually:
http://stackoverflow.com/questions/30131032/compile-opencv-with-tbb-on-raspberry-pi-2

Tried compiling latest Prusa edition and it worked!

eyal0, 2017-04-17:
Good to hear. I'm working on an update to the instructions.

eyal0, 2017-05-01:
@NiklasOlofsson Thanks for the link.  I wrote some instructions that should be easy to follow: https://github.com/OctoPrint/OctoPrint-Slic3r/wiki/How-to-install-Slic3r-on-RPi

closed 2017-05-01 by eyal0