OctoPrint/OctoPrint-Slic3r #17: slic3r command not working from Octoprint

kldavis4, 2017-05-03:
I just installed this plugin and everything seemed to go according to the instructions. When I attempt to slice a model from OctoPrint, though, it is not working. In the octoprint log I see:

`2017-05-02 21:29:06,684 - octoprint.plugins.slic3r - INFO - Running u'"home/pi/Slic3r/slic3r.pl" --load "/tmp/slicing-profile-temp-g6CJSS.profile" --print-center "52.000000,58.500000" -o "/tmp/tmpxoGUWl.gco" "/home/pi/.octoprint/uploads/trispinner1_test_1.stl"' in home/pi/Slic3r`

and in the slic3r plugin log I see:

`2017-05-02 21:29:06,683 ### Slicing /home/pi/.octoprint/uploads/trispinner1_test_1.stl to /tmp/tmpxoGUWl.gco using profile stored at /tmp/slicing-profile-temp-g6CJSS.profile`

In the OctoPrint UI I just see a modal that doesn't ever go away. The tmp gcode file is empty. I can manually run the command from the cli and it works fine to generate the gcode file. Any ideas what is happening

eyal0, 2017-05-03:
I'm able to reproduce this bug.  So far, I only see a problem when I set the "After Slicing" option to anything other than "Do Nothing".  Is that the same for you?

If so, as a workaround, set the "After Slicing" option to "Do Nothing", slice, and then print the object.

eyal0, 2017-05-03:
Oops, sorry, that was only because I didn't have a printer connected.  For me, slicing works and the modal goes away.

Maybe there is a bug in the javascript?  Can you try opening the developer's console and looking there?  On Windows or Linux, I think that you just press F12.  Maybe there is an error displayed there?

kldavis4, 2017-05-03:
I am not seeing anything in the Chrome dev console that looks like javascript errors. I have 'Do Nothing' selected as the option. Any other ideas for troubleshooting?

eyal0, 2017-05-03:
Is `### Slicing /home...` the last line that you see in the file?  When slicing completes, it should write to the log `### Finished...`.

Maybe you can try running octoprint with debugging turned on?  On OctoPi, from the CLI, type in:

```
killall octoprint
/home/pi/oprint/bin/octoprint -v
```

If you're not using OctoPi, you might need to do it differently.  This will log more information to the console and more to both `octoprint.log` and `plugin_slic3r_engine.log`.

Another place to look is the network tab in the Developer Tools.  You should see the command sent to octoprint right when you press "Slice".  The url should include the name of the file that you're trying to slice.  What is the Status Code?

kldavis4, 2017-05-03:
I just figured it out :( Somehow I left the leading slash of the path to the slic3r_engine. Not sure why octoprint wasn't giving me a meaningful error in the log, but looks like this is a non-issue. Thanks for your help and sorry for the bother.

closed 2017-05-03 by kldavis4

eyal0, 2017-05-03:
Glad to hear that you found the issue.  Now that you point it out, I see the missing leading slash in the log.

I'll see if I can improve the error reporting so that it doesn't trip up the next person!