OctoPrint/OctoPrint-Slic3r #19: Allow a standard threadcount

gappleto97, 2017-05-06:
It would be nice if I could have a box which always passes the --threads argument to slic3r. At the moment it *appears* to ignore this argument if I upload a profile. If you can point me at the correct files to change, then I'm more than happy to submit a PR.

eyal0, 2017-05-07:
From the slic3r documentation:

`    -j, --threads <num> Number of threads to use (1+, default: 12)`

So the default is 12, which seems like a good number!

I prefer not to add arguments like `--threads` into the Slic3r plugin because it's too difficult to keep up with all the changes that keep happening in Slic3r.  Instead, Slic3r is designed to work with whatever flags Slic3r supports.

So I see a couple options:
1. You could make a s script that would do it.  If you wrote this file:
   ```
   #!/bin/bash

   ./slic3r.pl --threads 100 $@
   ```
   and saved it named `slic3r_threads_100.sh`, made it executable, and then used that as your slic3r executable, that would work.
2. A second option is to change your settings in OctoPrint-Slic3r so that the executable is `slic3r.pl --threads 100`.  Then you wouldn't need to write a script.  However, this currently does *not* work in my testing.  If you want to write the PR for that, that would be useful!

What do you think?

eyal0, 2017-06-11:
I think that the workaround should be sufficient and the user has not responded.

closed 2017-06-11 by eyal0