OctoPrint/OctoPrint-Slic3r #20: Newbie installation help

oskarkroon, 2017-06-01:
Hi!

Followed the tutorial step by step and got this error message at the end, any ide what I am doing wrong? Probably a lot but anyway! ;)
__________________________________________
Raspberry Pi2

PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
8.0
0.13.0
______________________________________________________-
Octoprint ver. 
Version 1.3.2 (master branch)

_______________________________________________________
pi@octopi:~/Slic3r $ if [ ! -z "${TBB_DIR}" ]; then
> mkdir -p ${TBB_DIR}
> cd ${TBB_DIR}
> wget ${TBB_RELEASE}
> tar xzvf $(basename ${TBB_RELEASE})
> cd tbb*
> make tbb CXXFLAGS="-DTBB_USE_GCC_BUILTINS=1 -D__TBB_64BIT_ATOMICS=0"
> cd ${TBB_DIR}
> mkdir libtbb-dev_${TBB_VERSION}_armhf
> cd libtbb-dev_${TBB_VERSION}_armhf
> mkdir -p usr/local/lib/pkgconfig
> mkdir -p usr/local/include
> mkdir DEBIAN
>
> cd ${TBB_DIR}/libtbb-dev_${TBB_VERSION}_armhf/DEBIAN
> cat > control << EOF
> Package: libtbb-dev
> Priority: extra
> Section: universe/libdevel
> Maintainer: ${YOUR_NAME} <${YOUR_EMAIL}>
> Architecture: armhf
> Version: ${TBB_VERSION}
> Homepage: http://threadingbuildingblocks.org/
> Description: parallelism library for C++ - development files
>  TBB is a library that helps you leverage multi-core processor
>  performance without having to be a threading expert. It represents a
>  higher-level, task-based parallelism that abstracts platform details
>  and threading mechanism for performance and scalability.
>  .
>  (Note: if you are a user of the i386 architecture, i.e., 32-bit Intel
>  or compatible hardware, this package only supports Pentium4-compatible
>  and higher processors.)
>  .
>  This package includes the TBB headers, libs and pkg-config
> EOF
>
> cd ${TBB_DIR}/libtbb-dev_${TBB_VERSION}_armhf/usr/local/lib
> cp ${TBB_DIR}/tbb*/build/*_release/libtbb.so.2 .
> ln -s libtbb.so.2 libtbb.so
>
> cd ${TBB_DIR}/tbb*/include
> cp -r serial tbb ${TBB_DIR}/libtbb-dev_${TBB_VERSION}_armhf/usr/local/include
>
> cd ${TBB_DIR}/libtbb-dev_${TBB_VERSION}_armhf/usr/local/lib/pkgconfig
> cat > tbb.pc << EOF
> # Manually added pkg-config file for tbb - START
> prefix=/usr/local
> exec_prefix=\${prefix}
[... 519 lines cut ...]
  Parse errors: Bad plan.  You planned 8 tests but ran 0.
t/slice.t                    (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: No plan found in TAP output
t/speed.t                    (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 2 tests but ran 0.
t/support.t                  (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 28 tests but ran 0.
t/svg.t                      (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 2 tests but ran 0.
t/thin.t                     (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 23 tests but ran 0.
t/threads.t                  (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: No plan found in TAP output
t/vibrationlimit.t           (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 9 tests but ran 0.
Files=34, Tests=0, 88 wallclock secs ( 2.25 usr  0.61 sys + 69.15 cusr  7.65 csys = 79.66 CPU)
Result: FAIL
Some tests failed. Please report the failure to the author!

eyal0, 2017-06-07:
Hmm... Maybe the master branch is broken?  Try:

    SLIC3R_VERSION="version_1.34.0"

And then do the last three lines again, from `git checkout ...` onward.

oskarkroon, 2017-06-09:
That worked just fine, did the whole process again and just changed it to a different folder and it passed this time!

eyal0, 2017-06-11:
Glad to hear it!

closed 2017-06-11 by eyal0