OctoPrint/OctoPrint-Slic3r #22: Slic3r not in Slicer selector

shguro, 2017-07-31:
Hi,
i have installed the Slic3r and the Plugin on my PI but the Slic3r will not be shown in the Slicer selector.

Thanks for Help

eyal0, 2017-08-02:
Can you explain more?  Maybe a screenshot?

leobel96, 2017-08-04:
I had the same issue
[image] I resolved disabling cura slicer from plugin manager and rebooting raspberry

eyal0, 2017-08-07:
This can happen if the directory that you put for OctoPrint-Slic3r is wrong.

Is it working for you now?

leobel96, 2017-08-07:
The directory was correct. I resolved disabling cura slicer

eyal0, 2017-08-07:
That's strange.  You should not need to disable cura for slic3r to work.  They should be able to work together.

If you are satisfied, I will close this issue.

closed 2017-08-07 by eyal0