OctoPrint/OctoPrint-Slic3r #23: Prusa Slicer

ManuGithubSteam, 2017-08-01:
Hi

Why do newer versions of the prusa slicer not work correctly ?

Thanks

eyal0, 2017-08-02:
Which version did you try and what exactly isn't working?

ManuGithubSteam, 2017-08-02:
I tried the newest version from github but int cant compile because of libopengl/mesa version mismatch.
However i don't have enough Raspberry experiance to upgrade to the next debian version. in the pi. I dont even know if that is possible on ARM....

eyal0, 2017-08-02:
Are [these](https://github.com/OctoPrint/OctoPrint-Slic3r/wiki/How-to-install-Slic3r-on-RPi) the instructions that you tried?

Can you try those instructions first with version 1.34.1?  That version works for me.  I haven't tried a newer one.

After we see that 1.34.1 works, then try a later one.

These are the settings that  you need to use:

```
SLIC3R_REPO="https://github.com/prusa3d/Slic3r.git"
SLIC3R_VERSION="version_1.34.1"
SLIC3R_DIR=~/Slic3r
TBB_RELEASE="https://github.com/01org/tbb/archive/2017_U5.tar.gz"
TBB_DIR=~/tbb
TBB_VERSION="5"
YOUR_NAME="Your Name"
YOUR_EMAIL="email address"
```

eyal0, 2017-08-07:
Did it work?

eyal0, 2017-08-31:
Long time no reply.  Closing.

closed 2017-08-31 by eyal0