OctoPrint/OctoPrint-Slic3r #24: [Request] Select printing temperature during slicing

leobel96, 2017-08-10:
Hello. Is there a way to choose filament for slicing? I saw that in Octoprint's Settings I can add filaments but this is only for heatbed and nozzle preheating and not also for printing. Should I upload a different printing profile for EVERY filament I use? Thank you

eyal0, 2017-08-31:
Hmm, can you explain more?  Where in OctoPrint's settings are you able to select a filament?  Can you upload an image?  Maybe it's a new feature in OctoPrint and I haven't seen it before.

For slicing, the only thing that matters about the filament is the thickness and temperature.  To change thickness you'd usually need to change your print head which is a big operation.  So the only thing that you might change is temperature, perhaps from ABS to PLA or vice-versa.

I can look into it.  Please provide a screen shot or more details about the filament selection menu, thanks!

leobel96, 2017-08-31:
In "temperature" tab of octoprint's configuration you can add filaments and change temperatures. Here you can see the temperature tab https://github.com/foosel/OctoPrint/wiki/Configuration. From the main tab of octoprint you can, then, select the filament you have previously set and the heatbed and nozzle will heat until they reach the temperatures you have set. Unfortunately when you print the temperatures change according to the temperatures indicated in the G-CODE file generated by the slicer. So there should be an option to choose slicing temperature into slicer.

eyal0, 2017-09-02:
I think that @foosel 's answer is correct: If you want to just use the Slic3r plugin, you should make a profile per temperature.  But if you want to adjust it depending on filament then you should use the [Slicer plugin](http://plugins.octoprint.org/plugins/slicer/).

Anyway, you probably don't want to just adjust temperature when you change filament.  You probably would also need to adjust fan settings and cooling.  Supporting the myriad of possibilities would be too difficult.  You can make one profile per filament.

closed 2017-09-02 by eyal0

leobel96, 2017-09-02:
Perfect. Is It normal that if I try to upload a Slic3r Prusa edition profile on the Slic3r plugin I obtain an error? Am I obliged to install the Prusa edition version inside Octoprint?

eyal0, 2017-09-03:
No, you can upload any version of profile, it should work.

Make sure that you upload a regular profile and not a "bundled" profile.  Your uploaded profile should not have any lines like "[filament:" in it, for example.