OctoPrint/OctoPrint-Slic3r #25: Profile description

Passaita, 2017-09-10:
I have installed and tested this plugin with Slic3r 1.33.8. Versions with TBB on RPI
Is it possible to add more descriptive profile description field?
Having multiple profiles with different variables not just filament (i.e. Infill, skirt, brim, support, speed and etc.) will become confusing at some point as the name of profile won't fit enough information.

As an alternative having control over some of these option through the web interface would be extremely beneficial.

eyal0, 2017-09-10:
This is the same as https://github.com/OctoPrint/OctoPrint-Slic3r/issues/24 .

I suggest that you get the Slicer plugin for octoprint.  It will let you set all the options that you've named in addition to letting you rotate the object, scale it, and put multiple on the platform.

You can get it here: http://plugins.octoprint.org/plugins/slicer/

closed 2017-09-10 by eyal0

Passaita, 2017-09-11:
Awesome! Thank you. Missed this plugin.