OctoPrint/OctoPrint-Slic3r #27: can't post a request, this is not an issue but a request

cnlson, 2017-12-21:
is there any way to make this pull slic3r prusa edition?

https://github.com/prusa3d/Slic3r

eyal0, 2017-12-26:
Did you try the instructions?

https://github.com/OctoPrint/OctoPrint-Slic3r/wiki/How-to-install-Slic3r-on-RPi

cnlson, 2017-12-26:
> Did you try the instructions?
>
> https://github.com/OctoPrint/OctoPrint-Slic3r/wiki/How-to-
> install-Slic3r-on-RPi

I'm looking to put a different slic3r, the prusa edition that has a
different git than the standard slic3r

eyal0, 2017-12-27:
I understand.  In the instructions, it explains how to put the prusa edition.  Did you try that?

eyal0, 2018-02-28:
It's been too long without an answer.  I'm closing this.  Re-open if you're still having trouble.

closed 2018-02-28 by eyal0