OctoPrint/OctoPrint-Slic3r #28: RPi Slic3r install instructions missing steps

rambo, 2018-01-16:
"prusa3d, latest known working" I had to run at least the following (I am not 100% sure they were the only steps I took) after failed compilations to succesfully compile.

`
sudo apt-get install libeigen3-dev libglew-dev libglewmx-dev cpanminus
cpan App::cpanminus
cpan ExtUtils::Typemaps::Basic
cpan cpan ExtUtils::XSpp::Cmd
sudo chown -R pi:pi /home/pi/Slic3r/local-lib/
`

A few missing libraries (eigen, GLEW) and some missing Perl dependencies that Build.pl does not handle for some weird reason. And I guess the `--sudo` argument makes at least parts of the local lib owned by root subsequent builds of for example Slic3r::XS module will fail because the module cannot be installed with normal user privileges.

rambo, 2018-01-16:
And this was a fresh OctoPi image before I went messing with it.

eyal0, 2018-01-16:
Could be that the latest needs different libraries.  It happens.

Would you add those lines to the wiki?

https://github.com/OctoPrint/OctoPrint-Slic3r/wiki/How-to-install-Slic3r-on-RPi

`--sudo` is a problem, it's true, because the permissions get screwed up with root files in the local directory.  I wonder if it would be better to just to everything `sudo root`.  I mean, it's only an OctoPi, most people aren't too paranoid about the security of their printer controller.  If you think that it's best, you can change the `--sudo` line to a regular `sudo` run.  Up to you.

Thanks!

DongerZonie, 2018-05-02:
As of now you also need to install libcurl4-openssl-dev

eyal0, 2018-05-02:
It's a wiki, you can go ahead and edit it yourself if you like.  If not, I
can get around to it.

rambo, 2018-05-04:
On 03/05/2018 20.18, Kieran Ramnarine wrote:
> @rambo <https://github.com/rambo> did any of those need sudo-er? I'm
> going to add it to the wiki too
> 

I think some of the cpan commands did something weird with sudo and
thats why the later chown back to pi:pi

But it's been a while so I'm not sure what exactly it was.

/Rambo

eyal0, 2018-05-31:
The worry with an install script is that I don't want people blindly downloading scripts from the internet and running them.  It isn't safe.  But in the wiki isn't that much better, either.

I had a plan to have the script added to the OctoPi releases, that way it would be easy and reliable for users.  Never got around to it, though.

eyal0, 2018-07-04:
Seems to be working for people now.

closed 2018-07-04 by eyal0