OctoPrint/OctoPrint-Slic3r #30: Assistance Request

ianjolin, 2018-02-23:
Im trying to install but Im stuck at step 3.2 from (https://github.com/OctoPrint/OctoPrint-Slic3r/wiki/How-to-install-Slic3r-on-RPi). 
I cope and paste the script into my terminal window but nothing happens.

Any help would be appreciated.

eyal0, 2018-02-25:
> install-Slic3r-on-RPi).

Did you press enter after you copy and paste?

Can you send a log of the output?

Thanks.

reederda, 2018-02-28:
I am having the same issue- I've tried straight copy-paste, as well as creating a .sh file and running it from there. The .sh file output reports "Syntax error: end of file unexpected (expecting "fi")"

eyal0, 2018-02-28:
Ah, I see the problem.  There was a bad edit in the wiki that put too much whitespace.  Please try again @reederda and @ianjolin .  If it works, then please close this issue.  Thanks!

ianjolin, 2018-02-28:
Thank you for the assist.

eyal0, 2018-03-01:
I don't know about any of that. I'm just a volunteer that maintains the slic3r plug-in for octoprint. If my changes solved your installation problem, please close this issue.

If you have an issue with a different plug-in, you should find that github page and open a descriptive bug report.

ianjolin, 2018-03-01:
I’ll be home this evening and begin the install. I should know by tomorrow morning the latest.
Thanks again for the assist.

ianjolin, 2018-03-02:
THe solution worked perfectly.

closed 2018-03-02 by ianjolin