OctoPrint/OctoPrint-Slic3r #33: Help Needed

bdog1, 2018-03-05:
Do I have to perform all the steps listed at https://github.com/OctoPrint/OctoPrint-Slic3r/wiki/How-to-install-Slic3r-on-RPi in ADDITION TO just installing the slic3r plugin within OctoPrint? I installed the plugin and then was able to add the path to the directory that contained the slic3r engine within the SD card loaded in the Raspberry Pi3 . I get a message like failed to slice Error Code 2 or some such message.

bdog1, 2018-03-05:
I tried running the first command to increase the swap size and got 
[image]

bdog1, 2018-03-05:
Extra Characters after command?

eyal0, 2018-03-05:
The steps in the wiki let you install slic3r on the Rapsberry Pi.  If you have a Raspberry Pi compatible version of slic3r on an SD card, you can just give the path to that.  But not all slic3r engine files will be compatible, for example, if you grabbed a linux one, it won't work.  The procedure in the script will build one that works.

The `sed` command had a typo, it needed a space after the `-i`.  I added one and I think that solves the problem.  Please try again and let me know.

bdog1, 2018-03-05:
Thanks so much for the reply! 

I am running a Pi 3 and have installed Octopi sucessfully. I have also expanded the memory space on the card so that the entire capacity of the card can be used -16 Gig. To speed things  up it sounds like downloading a version of Slic3r to the SD card would be the way to go. 

Do the direction that you graciously provided on the Wiki cover that method?

Thanks again for the help.

eyal0, 2018-03-06:
Hmm.  It's a great idea but I'm not sure if it would work.  The Slic3r installation process installs a few different files and some libraries, too.  I'm not sure if it would be possible to just copy the man slic3r flies to a new system and run them because maybe all the libraries that were created along the way would be missing and it wouldn't run.  Worth trying!

Maybe I could try to put all the slic3r files in a zip file and see if they'll work when unpacked on a different system?  I'll look into it.

eyal0, 2018-03-11:
Can you send a log of the output? What was the error in the compilation?

bdog1, 2018-03-12:
I could send a log if I knew where to look and how to retrieve it. :)

eyal0, 2018-03-12:
Are you able to copy and paste all the text on the screen to a comment on github?

Or you could type:

`screen -L`

before you start and then

`exit`

when you finish and there will be a file created called screenlog.0 that has the log in it.

Or just take a photos of the screen with your phone?  :-)

bdog1, 2018-03-15:
I reinstalled Octopi fresh on a new 8 Gig Micro SD card since the 16 Gig I was using was too large for this purpose. I wont be storing a ton of .stl files upon it. Want to setup a skype session or something so that we can troubleshoot me installing the Slic3r fresh on the new Octopi installation?

eyal0, 2018-04-15:
Skype might be difficult as I'm not sure of your timezone.  Might not have much overlap.

Also, I don't have skype.

bayrakmehmet, 2018-05-13:
When i installing libextutils-cbuilder-perl on my raspberry pi it gives dependency error 

> The following packages have unmet dependencies: libextutils-cbuilder-perl : Depends: perl (>= 5.13.9) but it is not going to be installed E: Unable to correct problems, you have held broken packages.

But i already meets the needs here `perl -v`

This is perl 5, version 24, subversion 1 (v5.24.1) built for arm-linux-gnueabihf-thread-multi-64int (with 80 registered patches, see perl -V for more detail)

Could you please help me about the installing, if it is possible please share the cross compile instruction maybe i can compile slic3r using other linux distribution

eyal0, 2018-06-10:
I just installed Slic3r-1.39.2 successfully this morning.  Perhaps try the instructions again?  If you get an error, you can send the log or take a photo and put it here.

I didn't have any trouble installing `libextutils-cbuilder-perl`.  Maybe you can try `sudo apt-get update`?

Richardgriff, 2018-06-27:
Hi can you help,

I am following the instruction do build sli3r PE but i keep getting the following error when i try to compile the TBB

 Warning: conditional outside an IT block for Thumb.

This error occurs multiple times

eyal0, 2018-06-28:
Looks like it says that it's just a warning, though.  Is it making your compile fail?

Richardgriff, 2018-06-28:
AL looks like its not the TBB thats causing the slic3r to fail..

I have posted the CMake log files. If you can help under this issue i would be greatfull

[CMakeOutput.log](https://github.com/OctoPrint/OctoPrint-Slic3r/files/2146543/CMakeOutput.log)
[CMakeError.log](https://github.com/OctoPrint/OctoPrint-Slic3r/files/2146544/CMakeError.log)

eyal0, 2018-06-28:
That output looks really foreign to me.  I don't recognize it at all!  That's from when you run cmake?  Is it on RPi?

Richardgriff, 2018-06-29:
Yes this from the Cmake Logs on a RPi. 

I am trying the latest build 1.40 beta. Will try the 1.39.2 build tonight see if i get any different messages.

eyal0, 2018-06-29:
I know that there is a drive in Slic3r prusa edition to change the compile to use less perl and more c++.

 I tested recently with 1.39.2 and put the instructions on the wiki. Try those, see if it helps, and then we can figure out how to proceed.

Richardgriff, 2018-06-30:
Ok version 1.39.2 working but it took hrs I would say nearly 12. Not the 1-2 quoted lol.

I wont be doing that again.

Look at deploying a rasberry vm to do any further builds.

Also I would highlight stopping the octoprint service before compiling.

fMads, 2018-09-06:
I've tried install the latest release for Slic3r PE (version 1.41.0), but also got some errors.
CMake-logs are attached here.
[CMakeError.log](https://github.com/OctoPrint/OctoPrint-Slic3r/files/2356433/CMakeError.log)
[CMakeOutput.log](https://github.com/OctoPrint/OctoPrint-Slic3r/files/2356434/CMakeOutput.log)

I've followed the guide, and tried multiple times, and got to different points in the make process - not sure what to make of that...
In the last attempt (which got the furthest) I changed `sudo make -j8` to `sudo make -j2` - there I got to 94%.

I've got a RPi 3 Model B+

Maybe one of you can see anything from the logs? :)

eyal0, 2018-09-06:
I don't know how to fix that.  I don't have an RPi on which I could try to debug it, either.  Maybe you need to use 1.39.2 for now?  Or perhaps ask the Slicer PE developers?  They have a project on github, too.

roboknight, 2019-04-08:
It sounds like your compiles might be running out of memory.  Can't be sure, but I was getting weird compiler behavior compiling 1.3.0 (not the latest), and it seemed to fail at first.  Then I had to use the swap increase trick.  That seemed to fix things.  Maybe a little more swap?  Or don't run with -j2.  That just wastes memory on the rPi, since it probably can't efficiently do two compile steps at a time anyway.

eyal0, 2021-01-13:
Do it like this: https://github.com/OctoPrint/OctoPrint-Slic3r/issues/49#issuecomment-759109329

closed 2021-01-13 by eyal0