OctoPrint/OctoPrint-Slic3r #34: Slic3r fails, code 2

mchaos, 2018-04-08:
I installed the latest version of prusa slicer and I used the same version of prusa slic3r to make the profile, and it continually fails slicing.

eyal0, 2018-04-09:
Can you send a screenshot?  Did you configure the path to slic3r correctly?

mchaos, 2018-04-09:
I did configure it correctly and the test passed. I will try to get you a scrrenshot but may have to wait until later after work.

mchaos, 2018-04-09:
http://imgur.com/u7FAPO1
http://imgur.com/KDAN2vh

Let me know if there is anything else you want me to show.

mchaos, 2018-04-09:
Oh yeah, ignore the fact that the profile is names working, that was in anticipation of it working.. it didn't lol

eyal0, 2018-04-09:
You should try to telnet or ssh to the Raspberry Pi and run /home/pi/Slic3r/slic3r.pl and see what it says.

Also, you can look in /home/pi/.octoprint/logs/octoprint.log and look for more details on the error message.

mchaos, 2018-04-09:
I turned on logs. Hadn't checked yet. Good ideas about running it from cli.
That will have to wait until tonight. I don't have ssh open to the wild.
Let you know. Thanks.

mchaos, 2018-04-10:
I tried a clean compile.. this is the error I got.. "Can't locate Slic3r/XS.pm in @INC (you may need to install the Slic3r::XS module) (@INC contains: /home/pi/Slic3r/local-lib/lib/perl5/5.20.2/arm-linux-gnueabihf-thread-multi-64int /home/pi/Slic3r/local-lib/lib/perl5/5.20.2 /home/pi/Slic3r/local-lib/lib/perl5/arm-linux-gnueabihf-thread-multi-64int /home/pi/Slic3r/local-lib/lib/perl5/5.20.0/5.20.1 /home/pi/Slic3r/local-lib/lib/perl5 /home/pi/Slic3r/lib /etc/perl /usr/local/lib/arm-linux-gnueabihf/perl/5.20.2 /usr/local/share/perl/5.20.2 /usr/lib/arm-linux-gnueabihf/perl5/5.20 /usr/share/perl5 /usr/lib/arm-linux-gnueabihf/perl/5.20 /usr/share/perl/5.20 /usr/local/lib/site_perl .) at /home/pi/Slic3r/lib/Slic3r.pm line 58.
BEGIN failed--compilation aborted at /home/pi/Slic3r/lib/Slic3r.pm line 58.
Compilation failed in require at ./slic3r.pl line 16.
BEGIN failed--compilation aborted at ./slic3r.pl line 16.
"

eyal0, 2018-04-10:
That's a compile that failed.  Did you compiling following the instructions in the wiki?  Did you see a step in those instructions that failed?

Are you installing Slic3r Prusa edition or Slic3r regular edition.  I only really have experience with the former.

mchaos, 2018-04-10:
Prusa. No fail up until that point that I notice.

eyal0, 2018-04-10:
Can you send a log of the compilation?

mchaos, 2018-04-10:
> issuecomment-380006007
> ALDk8K8kwBWOqVsaN8b9YN6MzxZhHHk3ks5tnGK0gaJpZM4TLvxq

It looks like there are missing dependencies. I have already installed 2
things, that have gotten the process a bit further.

I am working through it. Ill let you know.

eyal0, 2018-04-10:
Let me know which dependencies you find are missing and we'll add them to the wiki.

mchaos, 2018-04-10:
Lol, I am not even sure at this point. I'll see if I can make a list. I can
go back in my history, but I have installed some things on a whim that I
didn't need and I don't think I will remember what worked

mchaos, 2018-04-10:
Okay, I got it working. I switched to the original slicer. Still had issues. Installed some stuff. Then I saw an error that said not enough room. Now I know I hadn't gotten that before because I was able to compile 2 different directories. But after I cleaned up some cache, it works. So. Have no clue what actually did it. But it is working.

eyal0, 2018-04-11:
Glad to hear that it is working now.  I haven't compiled the Prusa version in a while and maybe the directions have changed.

You ought to get OctoPrint-Slicer, too.  It works great with this plugin.  https://github.com/kennethjiang/OctoPrint-Slicer

closed 2018-04-11 by eyal0