OctoPrint/OctoPrint-Slic3r #35: When installing per Wiki....

sdernbach1970, 2018-05-29:
When I started to install the library, I get the following....

Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Reading package lists... Done
pi@octopi:~ $ sudo apt-get install -y git libboost-all-dev libboost-geometry-utils-perl libboost-system-dev libboost-thread-dev git-core build-essential libgtk2.0-dev libwxgtk2.8-dev libwx-perl libmodule-build-perl libnet-dbus-perl cpanminus libextutils-cbuilder-perl gcc-4.7 g++-4.7 libwx-perl libperl-dev libextutils-cppguess-perl libeigen3-dev libglew-dev libglewmx-dev cpanminus
Reading package lists... Done
Building dependency tree
Reading state information... Done
build-essential is already the newest version (12.3).
git is already the newest version (1:2.11.0-3+deb9u2).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libglewmx-dev : Conflicts: libglew-dev but 2.0.0-3 is to be installed
 perl-modules-5.24 : Breaks: libextutils-cbuilder-perl (< 0.280225)
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
pi@octopi:~ $

I am not sure what I need to do to resolve this, can someone please help?

Thanks,
Scott

jeffeb3, 2018-06-01:
I can confirm. I'm using a clean image 2018-03-13-raspbian-stretch-lite.zip and 2018-04-18-raspbian-stretch-lite.zip. They both have this issue from the clean image.

The libglew/libglewmx doesn't bother me as much (I will start by just installing the simple slic3r), but the perl-modules/extutils-builder-perl is pretty important, AFAICT. I don't know much about perl, or why these packages are having trouble.

I'm planning on trying this a bit more, but I don't know when I'll get a chance. I just wanted to confirm that it's not user error (unless we are a lot alike :) ).

jeffeb3, 2018-06-01:
I think this list might have been cultivated against jessie, not stretch. I'm working against this image, and I'm getting farther: http://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2017-07-05/

Cross your fingers!

eyal0, 2018-06-04:
> raspbian_lite-2017-07-05/

Great.  Please just update the wiki if you get any insights.  Thanks!

jeffeb3, 2018-06-04:
@eyal0 Sure. I still am not successful, but the cause of these issues is definitely that the list of apt repos doesn't work in stretch.

I was trying to run it in a CustomPiOS script, which mounts the image and runs a qemu emulator to pretend to be an arm processor. That was failing in the build script due to the "threads" library failing it's tests. So, instead, I tried just running these from a pi (running the Jessie image) and I'm getting the error that it's not able to find boost and that I should install it in a sane place, or set the environment BOOST_LIBRARIES. I am only trying to build the Alex version, I haven't tried the prusa edition, thinking the Alex version would be easier...

Long story short, I will update the wiki if I ever get there, but I also hope to have a repeatable way to build a raspberry pi image from this code when I'm done. I don't know when I'll get there though.

eyal0, 2018-06-10:
@jeffeb3 I just tried it, today, and it worked.  I modified the instructions slightly to run `make` with `sudo` because there was a copy command that was failing.  Other than that, everything worked great for me.

Do you want to try again?  Use the instructions to build 1.39.2 from the Prusa Slic3r build.

churusaa, 2018-06-23:
Installing in an updated Stretch Raspi image and can confirm that the instructions seem to prevent completion on this version of Debian for ARM because of an inability to install libextutils-cbuilder-perl since it wants a newer version of perl than what's in the repos.

'''pi@scoot:~/Slic3r-1.39.2 $ sudo apt install libextutils-cbuilder-perl
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libextutils-cbuilder-perl : Depends: perl (>= 5.13.9) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
'''
Installing perl6 makes no difference.

Managed to get ExtUtils CBuilder installed with cpan, at least:

'sudo cpan ExtUtils::CBuilder'

Still no joy.

eyal0, 2018-06-23:
I think that I saw this issue at some point but it went away, maybe because
of something that I did...

I remember installing curl from sources at some point but I don't know
which problem that solved.

If you can provide instructions for how you installed your version of
RaspPi image, I can try to do the same from scratch on a fresh sd card.

eyal0, 2018-06-24:
@churusaa 1-2 hours was a wild guess, maybe it was more.  I think that -j8 was excessive memory in my testing, too.  I'll perhaps change the instructions to -j4 ?

churusaa, 2018-06-24:
I have updated the install instructions to reflect the new requirements for Raspbian Stretch. Please let me know if you find any errors.

churusaa, 2018-06-29:
Has anyone else had a chance to test this on a fresh image and confirm that the steps work as documented? I suspect this should be resolved, and I can test this myself once my new SD cards arrive, but that won't be for a few more days.

DBa2016, 2018-07-04:
I installed it by the instructions on OctoPI/stretch, worked fine except the last step - "Now you can slice your stl files:". I just cannot find where to initiate slicing an STL file I uploaded...

eyal0, 2018-07-04:
@DBa2016 So it's working for you now?  You need to install the plugin, too, and set the filename in there, something like `/home/pi/Slic3r/slic3r.pl`.

DBa2016, 2018-07-16:
@eyal0 Yes, this works, I just had to reduce parallelism and it took a while. The slicer is not that great though, especially the lack of those three kinds of profiles is annoying (plus the inability to change the profile directly from the interface).

eyal0, 2018-07-16:
You ought to try the Slicer plugin, which will let you modify settings before printing.

BenchPressesBooks, 2019-04-18:
I am not able to get this to work with a clean OctiPi or Raspbian setup using the following directions:
https://plugins.octoprint.org/plugins/slic3r/

perl-modules-5.24 : Breaks: libextutils-cbuilder-perl (< 0.280225)

If I try to install libextutils-cbuilder-perl manually it does not work, and using apt-get says I need a newer perl (libextutils-cbuilder-perl : Depends: perl (>= 5.13.9)) but when I run perl -v it states i am running 5.24.1 which is odd. I am trying to work around this at this time.

eyal0, 2021-01-13:
Installing from the wiki is too hard.  Just download an AppImage as described in the README

closed 2021-01-13 by eyal0