OctoPrint/OctoPrint-Slic3r #40: Can't fint path to slic3r-console.

rgjertsen, 2018-10-16:
I am trying to set up the slic3r plugin, but it doen't work when I enter the slic3r-console path. 
The path I type in is "C:\Program Files\Prusa3D\Slic3rPE\slic3r-console.exe".
Is this correct? Is it because I use Slic3r PE and not ordinary Slic3r?

eyal0, 2018-10-16:
Oops, I meant to use forward single quote, not back single quote:

```
'C:\Program Files\Prusa3D\Slic3rPE\slic3r-console.exe'
```

Another thing to try is to open a terminal and make sure that the file really exists where you think it is.  Can you open a terminal (I think that you press the windows key and type "cmd" and then hit enter) and then type in one of those lines?

One final thing to try is to replace all backslashes (`\`) with forward slashes (`/`) because OctoPrint is written in python and might be having difficulty with the former.

After you get it working, I'll try to think of a way to make this process easier in the future.

rgjertsen, 2018-10-16:
I found the file in Terminal with double quotes, so the path is correct.
I tried entering the path in the plugin with both single and double quotes, and forward slashes, but nothing works. I just get a message that the path doesn't exist.
I'm all new to printing, so haven't done any of this before.

eyal0, 2018-10-16:
Hmm.  Let me think...  The computer where you are typing in that setting, is it the same as the one that is running OctoPrint?  Or are you using a Raspberry Pi that is connected to your printer that is running OctoPrint?

rgjertsen, 2018-10-16:
I have a Raspberry Pi 3 connected to my printer. And I am typing this in the browser on another computer with Slic3r PE on it.

eyal0, 2018-10-16:
Okay, there is lies the confusion.  The slic3r plugin runs on your Raspberry Pi and it is there that the slicing will be done.  So you need slic3r PE installed on your Raspberry Pi to use this plugin.  The advantage of using this plugin is that you can upload gcode files directly to the printer and the printer will slice and print them.  That's great for printing STIL files that you download online.

The drawback is that the slicing on the Raspberry Pi is a good bit slower than it is on your computer.  A fix that takes 60 minutes to print might take, like, 30 seconds to slice.  And you don't have as much control over the slicing as you do on your computer where you can visually move the model around, adjust it, etc.

If the idea of slicing on your Raspberry Pi still sounds like something you want to try, then you need this plugin.  Your next step is to install Slic3r on your Raspberry Pi.  The instructions are [here](https://github.com/OctoPrint/OctoPrint-Slic3r/wiki/How-to-install-Slic3r-on-RPi).  It will take a little while to complete the installation but you only have to do it once and it can run unattended.

Let me know if that helps.

rgjertsen, 2018-10-16:
Oooh, ok. Thanks for the help, man :)