OctoPrint/OctoPrint-Slic3r #42: Slic3r settings for existing profiles are not loaded in interface / Slicer tab missing

XReyRobert, 2019-01-14:
Using  Slic3r Prusa Edition  1.33.8 with TBB

Default profile is loaded, can be updated and save to a new profile. (save behaviour using the slice button or via the plugins settings

When trying to load back the new profile, I get "Unknown quality" "Unknown fill" etc... Advanced settings are not displaying the modded profile parameters but default ones. 

It means that one have to re-entre slicing parameters from default for each and every slice.

After checking, profiles are properly written to disk (.octoprint/slicingProfiles/slic3r) but never properly loaded.

eyal0, 2019-01-17:
Can you provide the log?  I will see if there are errors in there.

eyal0, 2021-01-13:
Re-open if this is still a problem.

closed 2021-01-13 by eyal0