OctoPrint/OctoPrint-Slic3r #44: Just tried to install the latest Slic3r PE, but it won't find to find WxWidgets 3.1

NovaViper, 2019-02-23:
I started trying to install v1.42.3, but it won't find WxWidgets 3.1. I currently 3.0.2 from using one of the older installs of Slic3r PE, but it doesn't seem to function with my settings (since I had Slic3r 1.42.3 on my laptop). Anyone know any sort of way to force install wxWidgets 3.1?

Goodfeat, 2019-02-26:
How did you compile 1.41.3?
I can't do it
The program requires the WxWidgets 3 library.
Error 255 is related to the config.ini file. I managed to get rid of this error on 1.39.2 by loading the configuration from the previous version of the slicer

NovaViper, 2019-02-26:
> How did you compile 1.41.3?
> I can't do it
> The program requires the WxWidgets 3 library.
> Error 255 is related to the config.ini file. I managed to get rid of this error on 1.39.2 by loading the configuration from the previous version of the slicer

I **didn't** get it to compile on 1.41.3, but got it to compile on 1.39.2

roboknight, 2019-04-09:
It's now on the Wiki, but there are a few things:
1) you can get rid of WxWidgets dependencies.  Edit the CMakeLists.txt file and comment out lines 26 and 28.  They are WxWidgets and GL related.  Unless you plan on using a monitor with OctoPi and doing slicing that way, you won't need them.
2) You won't be able to build it with the given Boost library.  rPi currently only has 1.62.  1.64 is needed at least.

NovaViper, 2019-04-09:
I'll give it try when I get home this weekend

roboknight, 2019-04-10:
So, I tried it.  The code still ends up with wxWidgets dependencies, even when I turn OFF the GUI in the options.  I'm working on a fix now, but it takes forever to finish compiling.

NovaViper, 2019-04-10:
Has the beta versions of 1.42 been tested yet? I started using that build because of the support for sequential printing

roboknight, 2019-04-12:
Well, I finally got it built... Probably hasn't really been tested enough, but it had some features I wanted to try out.  However, it won't work with OctoPrint just yet because instead of -print-center, its option is now just -center... Causes OctoPrint-Slic3r to barf for now.  But if you want to build it, I posted some info under one of the other issues.  Funny thing is, I actually started wanting to build 1.41.3, but looks like I ended up with 1.42.0-beta.

roboknight, 2019-04-13:
Okay, so you might want to check: #47

That is how I finally got things built.

eyal0, 2021-01-13:
> Well, I finally got it built... Probably hasn't really been tested enough, but it had some features I wanted to try out. However, it won't work with OctoPrint just yet because instead of [...]

I think that I fixed the `--print-center` vs `--center` thing already.

closed 2021-01-13 by eyal0