OctoPrint/OctoPrint-Slic3r #45: .xs failed

psyfor, 2019-03-03:
Trying to install slic3r and receiving the following error.  I an unable to fix based on a quick search on what other have done.  

src/libslic3r/Config.hpp: In member function ‘virtual bool Slic3r::ConfigOptionFloat::deserialize(std::__cxx11::string)’:
src/libslic3r/Config.hpp:68:29: error: cannot convert ‘std::basic_istream<char>::__istream_type {aka std::basic_istream<char>}’ to ‘bool’ in return
         return iss >> this->value;
                             ^~~~~
src/libslic3r/Config.hpp: In member function ‘virtual bool Slic3r::ConfigOptionInt::deserialize(std::__cxx11::string)’:
src/libslic3r/Config.hpp:127:29: error: cannot convert ‘std::basic_istream<char>::__istream_type {aka std::basic_istream<char>}’ to  bool’ in return
         return iss >> this->value;
                             ^~~~~
src/libslic3r/Config.hpp: In member function ‘virtual bool Slic3r::ConfigOptionPercent::deserialize(std::__cxx11::string)’:
src/libslic3r/Config.hpp:252:29: error: cannot convert ‘std::basic_istream<char>::__istream_type {aka std::basic_istream<char>}’ to  bool’ in return
         return iss >> this->value;
                             ^~~~~
src/libslic3r/Config.hpp: In member function ‘virtual bool Slic3r::ConfigOptionFloatOrPercent::deserialize(std::__cxx11::string)’:
src/libslic3r/Config.hpp:282:29: error: cannot convert ‘std::basic_istream<char>::__istream_type {aka std::basic_istream<char>}’ to  bool’ in return
         return iss >> this->value;
                             ^~~~~
error building src/libslic3r/PrintObject.o from 'src/libslic3r/PrintObject.cpp' at /usr/share/perl/5.24/ExtUtils/CBuilder/Base.pm line 174.
FAIL
! Installing ./xs failed. See /root/.cpanm/work/1551603194.9076/build.log for details. Retry with --force to force install it.
The XS/C++ code failed to compile, aborting
If everything was installed properly,you should be able to run Slic3r with the command ./slic3r.pl

gandlers, 2019-03-29:
yes, I am having exactly the same error message when following the instructions on the octoprint plugins page.
I had to install cpanminus for the slic3r_install.sh script to run, but it fails when it gets to XS.

I had a look at the CPAN build.log but that doesnt help me either:

```

cpanm (App::cpanminus) 1.7044 on perl 5.024001 built for arm-linux-gnueabihf-thread-multi-64int
Work directory is /root/.cpanm/work/1553865588.23346
You have make /usr/bin/make
You have /usr/bin/wget
You have /bin/tar: tar (GNU tar) 1.29
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by John Gilmore and Jay Fenlason.
You have /usr/bin/unzip
--> Working on ./xs
Entering /home/pi/Slic3r/xs
Checking if you have ExtUtils::Install 1.46 ... Yes (2.06)
Checking if you have Module::Build 0.38 ... Yes (0.4220)
Configuring /home/pi/Slic3r/xs
Running Build.PL
-> OK
Checking dependencies from MYMETA.json ...
Checking if you have Module::Build 0.3601 ... Yes (0.4220)
Checking if you have ExtUtils::Typemaps::Basic 1.05 ... Yes (1.05)
Checking if you have ExtUtils::ParseXS 3.18 ... Yes (3.35)
Checking if you have ExtUtils::Typemaps::Default 1.05 ... Yes (1.05)
Checking if you have ExtUtils::XSpp 0.18 ... Yes (0.18)
Checking if you have ExtUtils::Typemaps 1.00 ... Yes (3.35)
Checking if you have Test::More 0 ... Yes (1.302162)
Building and testing Slic3r-XS-0.01
-> FAIL Installing ./xs failed. See /root/.cpanm/work/1553865588.23346/build.log for details. Retry with --force to force install it.
(END)

```
I am trying to install slic3r 1.2.9 on octopi 0.16.0

Any one got any ideas?

eyal0, 2019-04-03:
The slic3r build is not very stable.  They sometimes add code that doesn't compile well.  Can you switch to a known, stable version?

roboknight, 2019-04-08:
Well, I fixed things this way on the rPi:

   cat /root/.cpanm/work/1554677050.28555/build.log  (your log will be elsewhere, but similar)
1) I looked at the build log... it appears the Boost development libraries aren't available on rPi?
   (Well, I guess they are, but OctoPrint's build doesn't have them).
2) Installed libboost-all-dev
   sudo apt-get install libboost-all-dev

There may be a better way to do this, but this is what I found worked.  And I was able to EVENTUALLY build the latest version of Slic3r.  However, I now can't seem to get the plugin to do anything...  but that's a different issue.

ALSO NOTE:  You *WILL* likely have to increase memory as he suggests.  However, you won't get any kind of "Out-of-memory" type errors.  They will likely look more like compiler bugs.

eyal0, 2019-04-08:
I'm glad to hear that it works.  Would you please update the wiki with the correct instructions?

As for getting the plugin to do stuff: Did you set the correct path in the slic3r plugin?  Also, you can look at the logs in `.octoprint/logs`, you might find your answer there.

roboknight, 2019-04-09:
Well, it appears that installing slic3r AFTER installing the plugin did something weird... Well, it boiled down to 2 things I think:
1) I was trying to run from the pi home directory.  I think this SHOULD be okay, but I noted that the other slicer uses /usr/local/bin ... so I ended up putting the slic3r there.  However, #2 was probably the real kicker...
2) I reinstalled the plugin.  That seemed to do the trick.  I don't know why, but it wouldn't let me do anything until I reinstalled the plugin.  I did try .octoprint/logs , but nothing ever ended up in them.  It was strange.

roboknight, 2019-04-09:
I would update the Wiki, but I'm not so sure that's a good idea with an old browser (I'm on my ancient macbook pro atm).  I may not be able to do much.  This version of Chrome can't be updated and its not supported here on github.

eyal0, 2021-01-14:
Don't bother to build slicer, just use an AppImage as described in the README.

closed 2021-01-14 by eyal0