OctoPrint/OctoPrint-Slic3r #46: Starting gcode not working properly

patricksmalley, 2019-03-12:
My start gcode doesn't seem to be working properly. 
The main, most noticeable issue is that my printer does not pause and warm up for my M190 and M109 commands. Instead, it will just start trying to extrude while cold.
============================================
;Generated from Rails.tmp.1552295660536.stl 7e020f439a8a002fd69e9ba398ecef966db105ac
; generated by Slic3r 1.1.7 on 2019-03-11 at 09:15:04

; perimeters extrusion width = 0.40mm
; infill extrusion width = 1.28mm
; solid infill extrusion width = 0.68mm
; top infill extrusion width = 0.68mm

G21 ; set units to millimeters
M107

;Basic settings: Layer height: {layer_height} Walls: {wall_thickness} Fill: {fill_density}
;Print time: {print_time}
;Filament used: {filament_amount}m {filament_weight}g
;Filament cost: {filament_cost}
G90									;Set absolute positioning
M107 								;turn off fan
G28 X0 Y0 Z0 						;home X, Y and Z axis end-stops
M190 S{print_bed_temperature} 	;Heat up the print bed
G0 Z10 F4000						;Raise up before probing.
G29 								;initiate z-probing
G0 X0 Y0 F9000 					;Go to front
M117 Cleaning...					;Put Cleaning message on screen
M109 S{print_temperature} 		;Uncomment to add your own temperature line
G0 Z0.3 							;Drop to bed
G92 E0 							;zero the extruded length
G1 X40 E25 F500 					;Extrude 25mm of filament in a 4cm line
G92 E0 							;zero the extruded length
G1 E-1 F500 						;Retract a little
G1 X80 F4000 						;Quickly wipe away from the filament line
G1 Z0.3 							;Raise and begin printing
============================================
Any ideas?

eyal0, 2019-03-15:
I don't know how this line works where there is a variable instead of a value:

M190 S{print_bed_temperature}

Maybe that's the issue?

patricksmalley, 2019-03-15:
> I don't know how this line works where there is a variable instead of a value:
> 
> M190 S{print_bed_temperature}
> 
> Maybe that's the issue?

Correct, I was using Cura variables. I forgot to change those over. 
Here are the Slic3r variables.
http://mauk.cc/mediawiki/index.php/Slic3r_placeholders
Thanks for the help!

closed 2019-03-15 by patricksmalley