OctoPrint/OctoPrint-Slic3r #47: Can't install Slic3R Pe on RPi

Baptou88, 2019-04-07:
Hello Everyone
I wish desperately to install Slic3r, Printrun and Octoprint on Raspbian, but impossible to install Slic3r Pe .
I followed the instructions of this link: [How to install Slic3r on RPi](https://github.com/OctoPrint/OctoPrint-Slic3r/wiki/How-to-install-Slic3r-on-RPi)

But from the second step I already have an error:
`sudo apt-get install -y git libboost-all-dev libboost-geometry-utils-perl libboost-system-dev libboost-thread-dev git-core build-essential libgtk2.0-dev libwxgtk2.8-dev libwx-perl libmodule-build-perl libnet-dbus-perl cpanminus libextutils-cbuilder-perl gcc-4.7 g++-4.7 libwx-perl libperl-dev libextutils-cppguess-perl libeigen3-dev libglew-dev libglewmx-dev cpanminus`

`Les paquets suivants contiennent des dépendances non satisfaites :
 libglewmx-dev : Est en conflit avec: libglew-dev mais 2.0.0-3 devra être installé
 perl-modules-5.24 : Casse: libextutils-cbuilder-perl (< 0.280225)
E: Erreur, pkgProblem::Resolve a généré des ruptures, ce qui a pu être causé par les paquets devant être gardés en l'état.`

And: 
` sudo make -j2`
`CMakeFiles/Makefile2:1389 : la recette pour la cible « src/libslic3r/CMakeFiles/libslic3r.dir/all » a échouée
make[1]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/all] Erreur 2
Makefile:160 : la recette pour la cible « all » a échouée
make: *** [all] Erreur 2`

Do you have solutions to these problems? Are there any newer tutorials?
thanks for the returns

eyal0, 2019-04-08:
I have not tried it in a long time.  That is a surprising error because I would expect that many people would have reported this if it were a common problem.  Perhaps try to remove libglew-dev from the list and see if it work?  Or to remove libglexmx-dev and see if it works?

If the first command doesn't work, I do not expect make to work either.

roboknight, 2019-04-08:
I think I got this too, but I pretty much ignored that error, hoping that I wouldn't care because I think it was for the GUI version of Slic3r?

roboknight, 2019-04-08:
Can't edit my previous comment... I downloaded the PE for Slic3r.  Since it uses cmake, you can comment out line 26 and 28 (one is for the GUI version, the other is for wxWidgets).  on the rPi, you likely won't need those.  You will still have to run:
sudo apt-get install libglew-dev  

by hand, but you shouldn't need the other one.  For some reason, the Slic3r developers didn't separate the GUI components from their code very well, so even removing the option doesn't seem to remove the dependancy.  Maybe it is something you should ask them to look into a bit for octoprint?  Or at least a leaner command line build?

Baptou88, 2019-04-08:
Thanks for the reply !

but what is the line 26 and 28 
@roboknight 
> you can comment out line 26 and 28

roboknight, 2019-04-09:
Sorry, it was late.  I meant line 26 and 28 of the CMakeLists.txt file.  It contains GUI support by default it appears.  HOWEVER... I have had trouble with the default 1.62 boost library that the Pi seems to come with.  I'm attempting to build 1.66.  1.62 doesn't seem to contain the Boost.Process headers.  You can download them, but for some reason 1.62 doesn't seem to get the support.  So I am attempting to build 1.66 to see if I can get that support.

eyal0, 2019-04-09:
It appears that Boost.Process was introduced in boost 1.64 https://www.boost.org/doc/libs/1_64_0/doc/html/process.html .

Seems like you're trying to install the latest Slic3r PE?  Would you want to first try and old version?

roboknight, 2019-04-09:
My browser isn't supported with github, so I can't edit my comments instead of adding new ones.  What I did was the following (In the Slic3r directory after I edited CMakeLists.txt):
mkdir build
cd build
cmake ..
make

I think I had pretty much everything installed.  You still need libglew-dev as above.  Note: I don't use -j2 on the rPi because the rPi just doesn't have the horsepower or memory to really run multi-process.  I've seen that around.  The -j option for make just allows make to run multiple processes to compile faster.  On your x86 beast, that's fine, but on the lil 'ole rPi... it'll likely just run you out of memory.

eyal0, 2019-04-09:
If you got it working, please modify the script in the wiki.  Thanks!

roboknight, 2019-04-09:
I have an older one... not for the Prussa, but I wanted to see a few of the newer features, like the 3D honeycomb.  If boost 1.66 fails, then I'll cave and give up on 3D honeycomb... I just wanted to see what that looked like :)

eyal0, 2019-04-09:
@roboknight I'm sure you want to see the new features!  How about if you compile boost sources from scratch?  It's not really that hard.  Download boost, unzip, then run:

```
./bootstrap.sh --with-libraries=process
./b2
sudo ./b2 install
```

It's actually pretty quick.  You'll need the correct list of libraries for bootstrap.  If the build complains about others, you need to add them.  Many of them are header-only so they don't require mentioning.

roboknight, 2019-04-09:
I'm in the middle of it.  It just hasn't finished (I'm building on the rPi... ugh)... Except I think you might have just saved me... I didn't add --with-libraries=process ...

roboknight, 2019-04-09:
I downloaded version 1.66 from the sourceforge... don't know if it is really the latest (I don't think so) At any rate, I ran into an issue with the --with-libraries... it keeps telling me its the wrong option --with-<option> process ... something like that.

roboknight, 2019-04-09:
I had to restart it without the option.  So I guess I'm pretty much getting all of boost.  I have a 16GB SDcard (probably what is slowing this down actually).  At any rate, it looks like it's "cup-o-coffee" time.

roboknight, 2019-04-09:
Looks like Boost 1.66 installed just fine.  I'm at 50%... much further than before.  So I'm hoping it will finish.

roboknight, 2019-04-09:
Okay, I'm unsure at this point, but I may have reached the compile limits of the rPi, no matter what.  It seems to be compiling one file, PlaceholderParser.cpp and just hanging... Its still there, but it doesn't seem like it has made any progress.  I'm going to leave it at this point and see if it finishes in the morning.

roboknight, 2019-04-09:
You may want to have a look at this when compiling Slic3r:
https://stackoverflow.com/questions/27588296/running-gcc-on-c-source-file-on-linux-gives-cc1plus-out-of-memory-allocati

The file I was getting stuck on may have a similar issue.  I basically ran "fromdos" on all the .cpp and .hpp files.  It cleared up the file I was getting stuck on.

roboknight, 2019-04-12:
Okay, I finally got the thing built... but I had to give up on *NOT* building the gui version... Apparently you only really get the gui_version no matter what.
```
sudo apt-get install wx-common
apt-get install libwxbase3.0-dev
sudo apt-get install libwxgtk3.0-dev
vi CMakeLists.txt 
```
When you edit the CMakeLists.txt file (in the main Slic3r directory)
```
option(SLIC3R_STATIC                    "Compile Slic3r with static libraries (Boost, TBB, glew)" ${SLIC3R_STATIC_INITIAL})
option(SLIC3R_GUI                       "Compile Slic3r with GUI components (OpenGL, wxWidgets)" 1)
option(SLIC3R_FHS                       "Assume Slic3r is to be installed in a FHS directory structure" 0)
option(SLIC3R_WX_STABLE                 "Build against wxWidgets stable (3.0) as oppsed to dev (3.1) on Linux" 1)
option(SLIC3R_PROFILE                   "Compile Slic3r with an invasive Shiny profiler" 0)
option(SLIC3R_PCH                       "Use precompiled headers" 1)
option(SLIC3R_MSVC_COMPILE_PARALLEL     "Compile on Visual Studio in parallel" 0)
option(SLIC3R_MSVC_PDB                  "Generate PDB files on MSVC in Release mode" 0)
option(SLIC3R_PERL_XS                   "Compile XS Perl module and enable Perl unit and integration tests" 0)
option(SLIC3R_ASAN                      "Enable ASan on Clang and GCC" 0)
option(SLIC3R_SYNTAXONLY                "Only perform source code correctness checking, no binary output (UNIX only)" 0)
```
I made a build directory in the Slic3r directory
```
mkdir build
cd build
cmake ..
make
```

This way, if things go wrong, you can blow away the build and start again.
I used Boost 1.66 built from source (as above in this thread).  It was pretty easy.  WxWidgets installs a lot of stuff, including GTX2.  Also, you will need to bump dphys-swapfile:
```
CONF_SWAPSIZE=3000
```
I did also run this BEFORE running cmake or make:
```
sudo apt-get install tofrodos
find .. -name \*.c -exec fromdos \{\} \;
find .. -name \*.cpp -exec fromdos \{\} \;
find .. -name \*.hpp -exec fromdos \{\} \;
find .. -name \*.h -exec fromdos \{\} \;
```
from the build directory.  The first command installs fromdos (you will likely have to install fromdos (or you can use SED and perform an equivalent procedure, but I don't know the command off-hand)) and the rest run fromdos on the source files.  It might help with the above issue.  It did help with one file that seemed to get "stuck".  You can try without doing this.  I don't know what is causing the issue here, but things just seemed to get stuck.  At any rate, hopefully you'll end up with slic3r-gui.  It still runs the same.  I don't know where the "GUI" part really is, but it appears to run and slice okay.

roboknight, 2019-04-12:
Oh, I almost forgot... before installing boost 1.66:
```
dpkg -S /usr/include/boost/version.hpp
sudo apt-get autoremove libboost1.62-dev
```
The first line determines which boost library you have installed.  I had libboost1.62-dev.  Then you want to remove that.  This way you'll have the 1.66 boost library.

roboknight, 2019-04-12:
Just to be clear, this was just a clone from the master branch... So I might actually be on whatever is after version_1.44.0-alpha4.  If you want 1.41.3, or another tag, you'll need:
```
git checkout 1.41.3
```
Or whatever version you want to checkout.

eyal0, 2019-04-12:
That's great!  The next step is to turn it all into one script so that it can be included in OctoPi.  If I had an RPi, I would do it!  But I don't have one for testing that I can use.

roboknight, 2019-04-12:
I was trying to figure that out because I'm still running into problems because some of the arguments have changed (I opened up an issue there...).. I'll clone the code from here and see if I have more luck... but I've got some other things to do before I get home.

roboknight, 2019-04-12:
So I believe I fixed my problem and I sliced my first file with Slic3r PE 1.42.0-beta.

https://github.com/OctoPrint/OctoPrint-Slic3r/issues/48

Has my change for __init__.py in OctoPrint-Slic3r, but it doesn't seem like the way to do it because if you are just using the regular slic3r now, it won't work.  I'm not sure what the best way to fix it is, but if someone wants to make the change, they must recompile the __init__.py:

```
cd /home/pi/oprint/lib/python2.7/site-packages/octoprint_slic3r
vi __init__.py    # you need to change line 280
python
>>> import py_compile
>>> py_compile.compile('__init__.py')
```
Then restart OctoPrint.

NovaViper, 2019-04-13:
How did you run that `tar xf boost_1_66_0.tar.bz2` command? It doesn't seem to do anything but let me continue writing stuff until I hit ctrl+z

eyal0, 2019-04-13:
You need to bunzip the file first.  So it's `tar xjf boost_1_66_0.tar.bz2`.

@roboknight Want to send a PR for the changes in init.py?  Is the fix in OctoPrint-Slic3r?  Which init.py?

Either have OctoPrint-Slic3r support both styles of flags, checking the version and doing the right thing.  Or, have an external script that wraps slic3r and converts the print-center to center as needed.

Baptou88, 2019-04-13:
I have install boost 1.66 and install: 
sudo apt-get install wx-common
apt-get install libwxbase3.0-dev
sudo apt-get install libwxgtk3.0-dev
but i still have the same issue
`xs/CMakeFiles/libslic3r.dir/build.make:902: recipe for target 'xs/CMakeFiles/lib            slic3r.dir/src/libslic3r/GCodeSender.cpp.o' failed
make[2]: *** [xs/CMakeFiles/libslic3r.dir/src/libslic3r/GCodeSender.cpp.o] Error             1
CMakeFiles/Makefile2:103: recipe for target 'xs/CMakeFiles/libslic3r.dir/all' fa            iled
make[1]: *** [xs/CMakeFiles/libslic3r.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
`

NovaViper, 2019-04-13:
I'm still getting that wxWidgets 3.0 error. Has anyone fixed it??

NovaViper, 2019-04-13:
> So I believe I fixed my problem and I sliced my first file with Slic3r PE 1.42.0-beta.
>
> #48
>
> Has my change for **init**.py in OctoPrint-Slic3r, but it doesn't seem like the way to do it because if [...]

How do I exit out of vi? It just has text but I can't save nor do anything but exit

roboknight, 2019-04-14:
Oh, yeah, you might want to use nano.  I'm a bit of a geezer when it comes to editors.  ':q' will get you out of vi.  That's COLON and the small 'q'.  

@NovaViper, have you installed the wxWidgets 3.0 libraries?  You need to have those.  Plus, I'm doing all of this on a rPi.  If that's what you're using, then you might want to check to see if wxWidgets 3.0 libraries are in your repos.
```
apt-cache search wx
```
will let you see all the things containing wx (which should be mostly wxWidgets things).

roboknight, 2019-04-14:
@eyal the latest tar usually can figure out when things are bzipped, so I stopped using the j.  But yeah, an older one will usually need to have it.  The wrapper script for slic3r-gui seemed like a good idea to me too.  The __init__.py was the one that ends up in the main site-packages directory.  That's the code that generates the arguments.  I didn't submit anything because I figured maybe there was a better way to create a wrapper script.  And that seems it would go with some kind of installer?  Maybe looking for slic3r-gui and if it finds it, then creating the necessary wrapper?

roboknight, 2019-04-14:
I did file #48, but maybe I need to file something differently?

NovaViper, 2019-04-14:
> Oh, yeah, you might want to use nano. I'm a bit of a geezer when it comes to editors. ':q' will get you out of vi. That's COLON and the small 'q'.
>
> @NovaViper, have you installed the wxWidgets 3.0 [...]

Yeah, I have wx3.0 and a couple of other wx stuff

roboknight, 2019-04-15:
```
sudo apt-get install wx-common
sudo apt-get install libwxbase3.0-dev
sudo apt-get install libwxgtk3.0-dev
```
It appears you have all of the 3.0-dev libraries.  If you have edited the CMakeList.txt file so that the "Build against wxWidgets (stable) 3.0" is selected (1 is true here), then the above libraries should work to fix the wxWidgets stuff.  You will likely need to completely clean the directory where you ran cmake.  I usually create a "build" directory (as I have above), so that if I need to re-run the cmake, I can just ```rm -rf *``` in that directory and re-run cmake with ```cmake ..``` .  You have all of the libraries, and if you've installed libboost-1.66, it should run on the standard OctoPrint distribution.

NovaViper, 2019-04-15:
Uh.. what's the name of the line to edit in CMakeList?

NovaViper, 2019-04-15:
Changed the setting, but still doens't work

NovaViper, 2019-04-15:
Well, I'm trying to the sudo make j2 command but PuTTY/KiTTY keeps on disconnecting from the pi hence the command never finishes. How can I make it stay connected??

roboknight, 2019-04-15:
I'm presuming you did ```cmake ..``` from the build directory before you tried to make anything.  Once you do that, you can just run ```make```.  The ```-j2``` will just waste memory on an rPi.  Additionally, if you didn't ```sudo apt-get install libwxbase3.0-dev``` it will fail.  I noticed I had a mistake above and forgot ```sudo```.  If the ```make``` does succeed, then you can run ```make install```.  That should just copy files where they need to go.  Although, I think I pretty much copied the binary to /usr/local/bin when I was finished.  I'll see if I can attempt a patch for the script that does most of this.  I haven't done that yet because I'm not familiar with the normal install script and haven't looked that closely at it.

roboknight, 2019-04-15:
Oh, also it helps if you just ```rm -rf *``` *IN YOUR BUILD DIRECTORY*.  I don't want you to accidentally blow something away that you need.

NovaViper, 2019-04-15:
It failed, saying Segmentation Fault

roboknight, 2019-04-15:
> I have install boost 1.66 and install:
> sudo apt-get install wx-common
> apt-get install libwxbase3.0-dev
> sudo apt-get install libwxgtk3.0-dev
> but i still have the same issue [...]

@Baptou88, I think this is the issue that I used ```fromdos``` for as well as increasing the swapfile size to 3000.  I think it is crashing here because it is running out of memory.  I have updated the original instructions to fix a couple of missing sudos and the ordering.

roboknight, 2019-04-15:
@NovaViper , You have an rPi 3 B+?  Do you know how much memory?

NovaViper, 2019-04-15:
I don't have know much memory I have and yeah, I have a Raspberry Pi 3 B+. And I did sudo make install before I did make

NovaViper, 2019-04-15:
Do I do all of this in the build directory??

roboknight, 2019-04-15:
Except for building libboost-1.66, it should be okay to do all of this from the build directory.  The instructions for resizing the partition can be done from there.  However, make sure you pay attention to the warnings because if you've already got a working OctoPrint setup, you might not be ready to lose that if things go horribly wrong with the partitioning.  It seemed pretty easy when I did it, but I *STARTED* there because I wanted to use the whole SDCard from the get go, so I didn't have much to lose except another DD to the SDCard.

NovaViper, 2019-04-15:
I've tried repartitioning and everything.. Still getting errors all over from make

roboknight, 2019-04-15:
@NovaViper, do you have the output from Make?

NovaViper, 2019-04-15:
Here: 

```
[ 35%] Built target glew
[ 35%] Built target nlopt
[ 35%] Built target admesh
[ 35%] Built target nowide
[ 35%] Built target clipper
[ 35%] Built target miniz
[ 35%] Built target glu-libtess
[ 35%] Built target polypartition
[ 35%] Built target poly2tri
[ 35%] Built target qhull
[ 35%] Built target semver
[ 35%] Building CXX object src/libslic3r/CMakeFiles/libslic3r.dir/pchheader.cpp.o

c++: internal compiler error: Segmentation fault (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-6/README.Bugs> for instructions.
src/libslic3r/CMakeFiles/libslic3r.dir/build.make:73: recipe for target 'src/libslic3r/CMakeFiles/libslic3r.dir/pchheader.cpp.o' failed
make[2]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/pchheader.cpp.o] Error 4
CMakeFiles/Makefile2:1427: recipe for target 'src/libslic3r/CMakeFiles/libslic3r.dir/all' failed
make[1]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

```

roboknight, 2019-04-15:
Yeah, that's the precompiled header.  You can "uncheck" the use precompiled headers.  That might help, but that one takes a TON of memory.  I didn't think to ask what other processes you might have had on the rPi.  I don't know if you have just the vanilla OctoPrint build.  Unchecking those guys will probably get you past this.  That sucker is HUGE... I had some real trouble on that one.  I don't know how many times I had to recompile that one because of issues.  You *MIGHT* also try making the swapfile 4000... but I eventually remember at that point running ```fromdos``` on the sources.  There is something in that precompiled header that ended up generating unicode, or something stupid and the preprocessor doesn't really handle it.  At any rate, try unchecking the precompiled headers first.  I had them checked, but maybe it is just better to do away with them.

NovaViper, 2019-04-15:
I'm not sure  how to do any of that.. <_>

roboknight, 2019-04-15:
@NovaViper, I updated the instructions a bit.  It more clearly points out the swap configuration, and a few other things (like the precompiled header option).

NovaViper, 2019-04-15:
More segmentation faults :/

roboknight, 2019-04-15:
@NovaViper, how far did it get?

NovaViper, 2019-04-15:
Got until the regular make command

NovaViper, 2019-04-15:
This is the log here: [ 35%] Building CXX object src/libslic3r/CMakeFiles/libslic3r.dir/pchheader.cpp.o

c++: internal compiler error: Segmentation fault (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See file:///usr/share/doc/gcc-6/README.Bugs for instructions.
src/libslic3r/CMakeFiles/libslic3r.dir/build.make:73: recipe for target 'src/libslic3r/CMakeFiles/libslic3r.dir/pchheader.cpp.o' failed
make[2]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/pchheader.cpp.o] Error 4
CMakeFiles/Makefile2:1427: recipe for target 'src/libslic3r/CMakeFiles/libslic3r.dir/all' failed
make[1]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

roboknight, 2019-04-16:
Okay, I sorted through the git differences in what I've done and there is one difference to what you likely have and what I have.  I didn't think it would matter, but maybe it does:

"src/slic3r/pchheader.hpp" line 95 of 187

I added 
```
#if SLIC3R_GUI
```
this is before all the ```#include <wx/....>``` lines.
then
"src/slic3r/pchheader.hpp" line 176 of 187

```
#endif
```
this is after all of the ```#include <wx/...>``` lines.
Basically, this put an #if - #endif block around all of the wx headers.  When I did that, my original intent was to remove them if I wasn't building the SLIC3R_GUI version.  However, I may
have unwittingly gotten what I wanted and reduced the size to compile on the rPi.  You are likely
hitting a hard compiler limit (which has been there for a while, but I don't know if it is still there in GCC 6.3.x, not the latest, but what I have on the rPi) that makes precompiled headers no more than 128M.  I believe this header is exceeding that.  What my original intent may have done is inadvertently removed the wx headers from the precompiled header file.  I don't have any ill effects, if this is truly what is letting me compile and *NOT* letting you compile.

NovaViper, 2019-04-24:
I've been scratching my head at this.. I still haven't been able to get it working

roboknight, 2019-04-25:
Is it the same error?  The other possibility is to "make" again.  Just make.  sometimes it keeps working through it and it eventually works because it gets by the thing causing the memory issues.

roboknight, 2019-04-25:
(Although, I will say, at this point, I haven't seen THAT much real difference from Slic3r PE and Slic3r 1.3.x)... Although I haven't sliced with a FEW of the features... but some of the features I think are only really useful if you use the GUI version, which I'm essentially not using.

NovaViper, 2019-04-26:
I haven't gotten to play with it since the last time i responded, but yeah, it's pretty much the same issue from last time

roboknight, 2019-05-14:
It didn’t build 3mf.cpp.o... not sure why, but you might have check for 3mf.cpp in the source.  Looking in that file might help.  I’ll see if I can find it in mine.

a2902793, 2019-10-04:
> ./b2
> sudo ./b2 install

will this result in duplicate files?

[2 more replies]

closed 2021-01-13 by eyal0