OctoPrint/OctoPrint-Slic3r #48: Slic3r Prusa Edition 1.42.0-beta

roboknight, 2019-04-12:
It appears the command --print-center is now just --center x,y

roboknight, 2019-04-12:
So on line 280 of __init__.py, I had to do the following to get slic3r-gui to work:
```
                args = ['"%s"' % executable, '--slice', '--load', '"%s"' % profile_path, '--center', '"%f,%f"' % (posX, posY), '-o', '"%s"' % machinecode_path, '"%s"' % model_path]
```
I had to add the --slice option, and --print-center changed to --center.  But it seems my configuration file worked okay.  I sliced my first file with OctoPrint and Slic3r PE 1.42.0-beta.

eyal0, 2019-07-26:
It will be hard to keep track of all the different flags that Slic3r has and changes. 😞

eyal0, 2021-01-13:
I think that I fixed this already.  Please check again.

closed 2021-01-13 by eyal0