OctoPrint/OctoPrint-Slic3r #49: How do I install PrusaSlicer 2.0.0?

NovaViper, 2019-05-30:
SO.. it's been a while since I tried to get this software working on my pi; now that Prusa has realized Prusa Slicer 2.0, I'm willing to give it a try again. But now since much has changed with the slicer.. how do I account the numerous changes made to it (especially the new name)?

sfjuocekr, 2019-06-11:
I manually installed PrusaSlicer from source, you will need an updated libboost as well.

The only problem is that PrusaSlicer changed some CLI parameters like center and the plugin fails to execute.

eyal0, 2019-06-12:
I haven't used it in a while so I won't be much use... If you tell me what things have changed, I can modify the code.  Or if you send a PR, I'll incorporate it.

NovaViper, 2019-06-12:
> I manually installed PrusaSlicer from source, you will need an updated libboost as well.
> 
> The only problem is that PrusaSlicer changed some CLI parameters like center and the plugin fails to execute.

How did you manage to do it?

sfjuocekr, 2019-06-18:
Removed libboost and dependencies, compiled the latest version and installed it.

Synced the latest PrusaSlicer from git, build it and installed it. 

Pointed the plugin to where PrusaSlicer put its stuff: /usr/local/bin/prusa-slicer

Changed __init__.py abit, it complained about two parameters:

Add --slice and change --print-center to --center.

It was pretty easy, but took days to compile... probably due to my slow SD card. I would recommend cross compilation on a beafy desktop or distcc to speed things up.

NovaViper, 2019-06-18:
Do you happen to have a copy of the changes you made?

eyal0, 2019-06-18:
I see a PR for it #50 but that will break older slic3rs. Can someone make a version that will also support older slic3rs?

sfjuocekr, 2019-06-25:
Yea I made a PR, but I did not take into account older slicers there :(

Maybe adding a simple checkbox for PrusaSlicer is the way to go here?

kaweksl, 2019-07-24:
Made some dirty script as workaround until someone gonna make true fix.
Instruction in comment
https://gist.github.com/kaweksl/204b71217deab3a92d8c1c536c96db1a

NovaViper, 2019-07-25:
> Made some dirty script as workaround until someone gonna make true fix.
> Instruction in comment
> https://gist.github.com/kaweksl/204b71217deab3a92d8c1c536c96db1a

How did you get the app image?

NovaViper, 2019-07-25:
> Made some dirty script as workaround until someone gonna make true fix.
> Instruction in comment
> https://gist.github.com/kaweksl/204b71217deab3a92d8c1c536c96db1a

Hey, how did you get it to run.. I keep on getting 'permission denied'

`
2019-07-25 11:38:34,179 stderr: /home/pi/PrusaSlicer/slicer.sh: line 11: /home/pi/PrusaSlicer/PrusaSlicer-2.0.0+linux64-201905201652.AppImage: Permission denied
2019-07-25 11:38:34,339 ### Finished, returncode 126
`

kaweksl, 2019-07-25:
You need to edit SLICER_PATH and set it to PrusaSlicer path

NovaViper, 2019-07-26:
I have the Rapsberry Pi 3B+, I've changed the path too to run the AppImage. How do I compile PrusaSlicer?

kaweksl, 2019-07-26:
Official PrusaSlicer AppImage won't work because Pi has for different cpu architecture.

Instructions are at PrusaSlicer github wiki https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md

Lately i was compiling it on Pi 4 and i have used this commands (I may forgot something )
It compiles with static libraries
```
sudo apt install git cmake build-essential libgtk2.0-dev libgl1-mesa-dev libglu1-mesa-dev 
# Gonna download to home folder
cd ~/
git clone https://github.com/prusa3d/PrusaSlicer.git
cd PrusaSlicer
# Selecting version (lateset at a time)
git checkout version_2.0.0
cd deps
mkdir build && cd build
cmake ..
# This gonna download and build static libraries
# you can't move or delete those after compiling, path gonna be hardcoded
# This gonna take a while
make
cd destdir/usr/local
pwd
# Save outputed path, gonna need it

# cd back to PrusaSlicer main folder
cd ~/PrusaSlicer
mkdir build && cd build

# change DCMAKE_PREFIX_PATH to path saved above
cmake .. -DSLIC3R_STATIC=1 -DCMAKE_PREFIX_PATH=/home/pi/PrusaSlicer/deps/build/destdir/usr/local
# This gonna take a while
make

#Test PrusaSlicer
/home/pi/PrusaSlicer/build/src/prusa-slicer -help
```

NovaViper, 2019-07-27:
@kaweksl I keep trying to run the make command but my terminal disconnects from the pi before it ever finishes. Is there a way to make it not do that?

kaweksl, 2019-07-27:
Is pi restarting ? what power supply you use, compiling need a lot of power, are you connecting pi via wifi or ethernet. Alternatively you could compile in "screen" or tmux session so you could reattach to terminal.

NovaViper, 2019-07-28:
@kaweksl I'm using the power supply that came with this kit I got, it's not the offical PSU though. But the pi never actually restarts. And I have the pi connected over the internet. I just hooked up my pi to my monitor and ran the command from there, though it did restart (that was because I accidentally turned off the switch that PSU has)

NovaViper, 2019-07-28:
Strange enough, even running the make command directly on the pi (this is when I run the command in PrusaSlicer/build) it gets stuck around 75% and nerves finishes. I let it run all night and it still didn't finish

eyal0, 2019-07-28:
It would be best to figure out how to cross compile it. The compile is too slow and not reliable!

NovaViper, 2019-07-28:
@eyal0 I have no clue how to do that though <_>

eyal0, 2019-07-29:
@NovaViper I'll try.  I need to get an RPi first for my testing.

eyal0, 2019-07-29:
@kaweksl Did you need to install TBB (Intel Threading Building Blocks) for your PrusaSlicer compilation to work?

kaweksl, 2019-07-29:
@eyal0 I think it is compiled by cmake in deps folder. Note that commands from above will build slicer with hard linked libraries that are independent from system libs but can't be moved after compiling slicer.

NovaViper, 2019-08-22:
Has anyone figured it out?

diabl0w, 2019-11-07:
@kaweksl how did you solve "undefined reference to `__atomic_compare_exchange_8'" when compiling the deps

diabl0w, 2019-11-07:
Just an update to this thread... I successfully built prusa with guidance from kaweksl's script above, but I also had to add "-DCMAKE_CXX_FLAGS=-latomic" to the openvdb section of 'deps-unix-common.cmake' (NOTE: I am running an octopi version that is running debian buster, not stretch).

Second comment, debian buster based octopi does have prusa-slic3r already in the repostories, but it is an old version (I think 1.39) and I had a few printing problems with that version. But if you are lazy, and still want prusa-slicer and are okay with an older version, then I think the easiest way to install it is to upgrade to debain buster and just install from the repos

NovaViper, 2019-11-09:
@diabl0w Could you detail how you got it to work?

diabl0w, 2019-11-09:
clone the latest prusa slicer then start kaweksl's steps at "cd deps" ... obv this should be CD PrusaSlicer/deps

NovaViper, 2019-11-09:
@diabl0w I clone it onto the pi right?

NovaViper, 2019-11-09:
@diabl0w How did you add the fix for openvdb?

NovaViper, 2019-11-09:
Also, I don't see openvdb in mine

diabl0w, 2019-11-09:
you need to be able to troubleshoot things like that yourself, use google and common sense... if it needs packages then install them... check prusa slicer github for any other system dependencies, actually I think kaweksl posted it in the first line of code

NovaViper, 2019-11-09:
> you need to be able to troubleshoot things like that yourself, use google and common sense... if it needs packages then install them... check prusa slicer github for any other system dependencies, actually I think kaweksl posted it in the first line of code

I've been looking online but haven't found anything to fix it. I also tried following everything with kaweksl and the PrusaSlicer wiki stated but none of that is working either. And each time im getting a *different* error each time

NovaViper, 2019-11-10:
I *still* can't get it to compile at all

```
gcc.compile.c++ bin.v2/libs/log/build/gcc-gnu-6.3.0/release/boost.locale.icu-off/link-static/threadapi-pthread/threading-multi/text_file_backend.o
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-6/README.Bugs> for instructions.
df^C^Z
[1]+  Stopped                 sudo make
```

sfjuocekr, 2019-12-30:
Sorry for being away for a while, if you SSH disconnects use a tool like screen or txmux.

I just followed the build guide for Slic3r and only manually updated libboost.

It took ages to compile on a RPi3, cross compiling should be much quicker but I was working remote anyway. If you are getting different compiler errors each time your RPi is probably getting too hot!

sfjuocekr, 2019-12-30:
Im betting this is now the easier option: https://github.com/koendv/prusa-slicer-raspberrypi/releases

NovaViper, 2019-12-31:
@sfjuocekr How do you cross-compile?

NovaViper, 2020-01-09:
@sfjuocekr Also, there's a few problems with that installation.. mainly it runs on the pi 4, which I don't have (I have the 3B+)

sfjuocekr, 2020-04-15:
Ok, I've dusted off my printer and decided to build the latest release.

It is building now, you need to edit the CMakeLists.txt to require CMake 3.7 instead its on the first line and I ditched the other dependencies it couldn't find and I couldn't get working so fingers crossed if it finishes.

NovaViper, 2020-05-06:
@sfjuocekr Did you ever get to compile?

Bougakov, 2021-01-12:
I seem to have found an easier solution. There is an AppImage of PrusaSlicer 2.3 built for armhf and armhf64 - https://github.com/davidk/PrusaSlicer-ARM.AppImage/releases 

I've downloaded it and extracted it using this - https://superuser.com/a/1507673/395907 - bash script.

```
root@orangepizero:/tmp# wget https://github.com/davidk/PrusaSlicer-ARM.AppImage/releases/download/version_2.3.0/PrusaSlicer-2.3.0-armhf.AppImage
root@orangepizero:/tmp# nano appi.sh
root@orangepizero:/tmp# chmod +x appi.sh 
root@orangepizero:/tmp# ./appi.sh -a PrusaSlicer-2.3.0-armhf.AppImage 
[..skipped extraction log..]
root@orangepizero:/tmp# cd ./PrusaSlicer-2.3.0-armhf/usr/bin/
root@orangepizero:/tmp/PrusaSlicer-2.3.0-armhf/usr/bin# chmod +x prusa-slicer 
root@orangepizero:/tmp/PrusaSlicer-2.3.0-armhf/usr/bin# ./prusa-slicer -h
PrusaSlicer-2.3.0+UNKNOWN based on Slic3r (with GUI support)
https://github.com/prusa3d/PrusaSlicer

Usage: prusa-slicer [ ACTIONS ] [ TRANSFORM ] [ OPTIONS ] [ file.stl ... ]
```

eyal0, 2021-01-13:
@Bougakov Did you actually need to unpack that AppImage?  I think that you ought to be able to just `chmod +x` on it and then run it directly.  That's how the images on Prusa's website work:  https://www.prusa3d.com/prusaslicer/

closed 2021-01-13 by eyal0

sfjuocekr, 2021-01-15:
Sorry, had not bothered recompiling again and kinda forgot about it :(

sfjuocekr, 2021-01-15:
g++: internal compiler error: Killed (program cc1plus)

Probably means your raspberry ran out of memory, I can recommend to hoop up a fast USB stick or SSD to add some swap as you will need it!

commit b4054cb2 (2021-01-13): Add instructions on how to get PrusaSlicer 

This fixes #49