OctoPrint/OctoPrint-Slic3r #54: Cannot build, stuck on make -j2

a2902793, 2019-10-04:
I was building Slic3r on my pi3 (followed the steps on wiki), where I got to `sudo make -j2` and it shows this error

[image]

Any suggestions on how can I solve this issue? Thanks.

eyal0, 2019-10-04:
Which version of slic3r are you using?

a2902793, 2019-10-06:
I followed the instructions,
> SLIC3R_REPO="https://github.com/slic3r/Slic3r.git"
> SLIC3R_VERSION="master"
> SLIC3R_DIR=~/Slic3r

so it's 1.3.0.

------
However, it says in the README that `Slic3r's stable versions 1.1.7 and 1.2.9`, so I gave 1.2.9 a try, checked it out and tried again, but this time I can't even get pass `perl Build.PL --sudo`, it throws errors like this

[image]

I've tried solving that error and [this reply in stackoverflow](https://stackoverflow.com/a/38659160/8959163) suggests that type casting needs to be done. So again, I tried adding `(bool)` to the lines showing errors and it built without errors but make files weren't generated (in which I saw this was fixed in Slic3rs commit [6e5938](https://github.com/slic3r/Slic3r/commit/6e5938c8330b5bdb6b85c3ca8dc188605ee56b98))

I'm facing problems and errors similar to [this issue](https://github.com/slic3r/Slic3r/issues/1690), should I reopen this issue onto Slic3r's page?