OctoPrint/OctoPrint-Slic3r #57: [Feature Request] Python 3 Compatibility

DrJuJu, 2020-08-10:
Please add Python 3 Compatibility as outlined [here](https://docs.octoprint.org/en/master/plugins/python3_migration.html).

kay102, 2021-01-10:
Hi,

Are there any actions ongoing towards Python3 for this plugin?

Kind regards

eyal0, 2021-01-10:
I haven't been supporting this much because I don't have a 3D printer anymore.  If someone else wants to send the PR, I can review it!

eyal0, 2021-01-10:
I can try but I'm not sure if I'd have a good way to test it.  I'll try to make time for this and make a new version.  Hopefully someone can test it out for me.

kay102, 2021-01-11:
@eyal0: When you say "test", do you mean try out if it works? I could do that...

eyal0, 2021-01-13:
Fixed by version 1.2.1

Let me know if this helps.  Thanks.

closed 2021-01-13 by eyal0

kay102, 2021-01-13:
Works fine for me. Thank you!