OctoPrint/OctoPrint-Slic3r #60: On a side note: it would be very helpful, if `plugin_slic3r_engine.log` contained the raw command with all the parameters.

eyal0, 2021-01-13:
On a side note: it would be very helpful, if `plugin_slic3r_engine.log` contained the raw command with all the parameters.

_Originally posted by @Bougakov in https://github.com/OctoPrint/OctoPrint-Slic3r/issues/59#issuecomment-759144818_

eyal0, 2021-01-13:
You must set the logging level in the Settings under "Logging":

[image]

Reopen this bug if that didn't help you.

Bougakov, 2021-01-13:
I have it enabled. My suggestion was to put the _raw command_ used to launch the slicer in the log "as-is".

Currently, only this line is saved there `2021-01-13 04:57:10,597 ### Slicing /home/octoprint/.octoprint/uploads/xyzCalibration_cube.stl to /tmp/tmpLBSqvb.gco using profile stored at /tmp/slicing-profile-temp-wYKe2B.profile`. If it was there I could have spotted the obsolete launch parameter myself without bothering you :) 

[image]

eyal0, 2021-01-13:
Under logging levels, where it says "Name", I see that it says "octoprint".  In my image above, you can see where it says "octoprint.plugins.slic3r".  Can you try to add a line that will have that?

Also, it's a little screwy how it works.  After you set the name and the level, you then need to press the blue plus sign on the right for that row to be added.

Please try that and let me know how it goes.

Bougakov, 2021-01-13:
oh, that's a really, really poor UX design choice on the Octoprint's side. Thanks for explaining where to look. Logging of the command indeed works:

```
2021-01-13 15:22:57,130 ----------------------------------------
2021-01-13 20:55:24,595 ### Slicing /home/octoprint/.octoprint/uploads/xyzCalibration_cube.stl to /tmp/tmpzq_saI.gco using profile stored at /tmp/slicing-profile-temp-dkOIiq.profile
2021-01-13 20:55:26,710 Running Prusa Slic3r >= 2.3
2021-01-13 20:55:26,712 Running u'"/usr/bin/prusa-slicer" -g --load "/tmp/slicing-profile-temp-dkOIiq.profile" --center "0.000000,0.000000" -o "/tmp/tmpzq_saI.gco" "/home/octoprint/.octoprint/uploads/xyzCalibration_cube.stl"' in /usr/bin
2021-01-13 20:55:32,535 stdout: 10 => Processing triangulated mesh
2021-01-13 20:55:32,537 stdout: 20 => Generating perimeters
2021-01-13 20:55:32,538 stdout: 70 => Infilling layers
2021-01-13 20:55:32,539 stdout: 30 => Preparing infill
2021-01-13 20:55:32,539 stdout: 88 => Generating skirt
2021-01-13 20:55:32,540 stdout: 88 => Generating brim
2021-01-13 20:55:32,541 stdout: 90 => Exporting G-code to /tmp/tmpzq_saI.gco
2021-01-13 20:55:32,541 stdout: Slicing result exported to /tmp/tmpzq_saI.gco
2021-01-13 20:55:32,543 ### Finished, returncode 0
2021-01-13 20:55:33,951 Analysis found in gcode: None
2021-01-13 20:55:33,953 ----------------------------------------
```

resulting g-code:

```
;Generated from xyzCalibration_cube.stl (hash: d316cc8422b4622151bc32fe2910f6a8731736bb)
; generated by PrusaSlicer 2.3.0 on 2021-01-13 at 17:55:28 UTC
```

eyal0, 2021-01-13:
I have filed a fix with the the OctoPrint project.  Here's how it will work if everything is accepted (look at the section called "Screenshots": https://github.com/OctoPrint/OctoPrint/pull/3957

I think that it's a less clunky interface.  Also, I think that it would have been impossible for you to have made the error that you made.  Don't you agree?

I'll close this for now.  Re-open if you see more difficulties.

commit 80d4de5f (2021-01-13): Use self._logger for logging

This fixes #60
closed 2021-01-13 by eyal0