OctoPrint/OctoPrint-Slic3r #61: Feature request: pre-select the slicer in the drop-down

Bougakov, 2021-01-13:
This is a very, very, very minor thing, but it is a bit annoying. Slicing dialog requires the user to select the (only) slic3r profile from the drop-down menu each time. Can this dialog be tweaked so it is selected by default? Or, maybe, let's get rid of this drop-down completely, as there can't be multiple slicers with this plug-in? 

Thanks!

[image]

eyal0, 2021-01-13:
It's a valid point and I have noticed this, too.  First, I'll describe the behavior that I know and you let me know if it matches what you're seeing.  Then we can talk about a solution.

The item that you have encircled with a red box is not the "slicing profile", it's the "slicer".  I know of two slicers for OctoPrint so far:

1. This one, OctoPrint-Slic3r
2. A cura slicer that is here: https://github.com/OctoPrint/OctoPrint-CuraEngineLegacy

When you add either of those, the behavior of OctoPrint changes.  First of all, it lets you upload stl files.  Without any slicers available, you cannot.

Second, when you upload an STL file, it will have a little magic wand next to it.  When you click on that, the above dialog opens.  I didn't make that dialog, it's part of OctoPrint's source code.

OctoPrint doesn't know which slicer you want to use so it'll just make you pick one.  **That's annoying!**

After you do pick one, however, the slicing profile will get selected automatically.  The profile that is selected is the one that has a star on it in the slic3r configuration.  That's your default slicing profile.

Does all that make sense so far I hope?

## Solution

If there is only one slicer installed, OctoPrint ought to just pick that one when it opens the dialog.  That'll save you one click.  You'll still be able to modify the rest of the options before you hit "Slice".

---

How does that sound to you?

eyal0, 2021-01-13:
Oh how annoying.  There *is* a setting for this but as far as I can tell, there is no way for you to set it without editing a file.  The edit is easy to do though:

open up ~/.octoprint/config.yaml

Look for:

```
slicing:
    defaultProfiles:
        slic3r: pretty_cfabs_v1
    defaultSlicer: slic3r
```

See how I put defaultSlicer in there?  Do that.  You might need to make that change while octoprint isn't running, I'm not sure.

eyal0, 2021-01-14:
I set a fix but this won't work for you until https://github.com/OctoPrint/OctoPrint/pull/3958 is merged into OctoPrint.  I'm not sure how long that will take.  Once that fix goes in then my fix will start working automatically.

Because there are no more actions needed from me here, I'll leave this closed.

PR #62 (2021-01-14): Add a button to set the default slicer
This fixes #61
commit b2a1e240 (2021-01-14): Add a button to set the default slicer

This fixes #61
closed 2021-01-14 by eyal0