OctoPrint/OctoPrint-Slic3r #64: Instructions on Octoprint site no longer seem to work.

dalmuti509, 2021-01-15:
The provided instructions on the octoprint site don't seem to work on a new install of the latest Octopi image.   

I suggest adding:
`sudo apt-get update`  into the script.

As well, some of the references don't seem to exist anymore (g++-4.7 and libwxgtk2.8-dev at least, but there might be more that I missed).

I tried finding how to fix this, but could not find the correct resources to do so.    I worked around the issue by using your newer recommendation from the Github wiki, but it still failed to run correctly. 

version 1.1.7 warns Running Slic3r under Perl >= 5.16 is not supported nor recommended
version 1.2.9 throws a bunch of errors.

eyal0, 2021-01-15:
I haven't tried building slic3r in a long time.

I know that for PrusaSlicer there are AppImage files which you can just download and run, no compilation necessary.  So if you're willing to use PrusaSlicer, do that.  The instructions are in the README.

If you really want slic3r, also check the wiki.  I think that the instructions are more up-to-date.

And if you figure it out, please update the wiki.

masonh, 2021-01-30:
I just installed slic3r from apt on my pi 2b after failing to build and failing to run the PrusaSlicer appimage. It is the only method that has worked for me, and the easiest. I added a note at the top of the wiki page so that others hopefully won't need to go down this rabbit hole like I did.

```
pi@octopi:~$ apt search slic3r
[...]
slic3r/stable,now 1.3.0+dfsg1-3 armhf [installed]
  G-code generator for 3D printers

slic3r-prusa/stable 1.39.2+dfsg-1+b2 armhf
  G-code generator for 3D printers
```