OctoPrint/OctoPrint-Slic3r #70: Won't Enable/No slic3r option in plugin list

solo2424, 2022-07-20:
I am really hoping to get some assistance with getting slic3r installed and running on OptoPrint. I installed the plugin using the plugin manager and appears to install with no issues. After installation, I restart OptoPrint but slic3r is not showing up in the left-hand panel under plugins. When I go back to the plugin manager, it is disabled.  I enable it but it still does not show up in the plugin list on the sidebar. After restarting, the plugin is once again disabled. 

I have installed slic3r on the raspberry pi using this [tutorial](https://community.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspberry-pi-os-debian/2337). Any assistance will be greatly appreciated! Thank you!

roy-stewart, 2022-07-26:
Hi @solo2424, it sounds like you may be running into a python version issue. Do you see this lightning bolt icon next to the plugin on the plugin manager screen?
[image]
If so, that means that the python version used with your OctoPrint install doesn't support OctoPrint-Slic3r.

solo2424, 2022-07-31:
That does seem to be the problem. Do you know what version I need and how to update? I imagine it is as simple as an apt-get command?

solo2424, 2022-07-31:
I was able to check and it shows I have two versions of python installed. Both within the requirements from OctoPrint-Slic3r (Python: >=2.7,<4)

pi@octopi:~ $ python -V command
Python 2.7.16
pi@octopi:~ $ python3 --version
Python 3.7.3

cp2004, 2022-08-25:
The problem is below:
```
2022-08-25 17:11:23,498 - octoprint.plugin.core - ERROR - Invalid syntax in plugin file of plugin slic3r
Traceback (most recent call last):
  File "C:\Users\charl\Development\OctoPrint Core\OctoPrint\src\octoprint\plugin\core.py", line 782, in _parse_metadata
    return parse_plugin_metadata(self.location)
  File "C:\Users\charl\Development\OctoPrint Core\OctoPrint\src\octoprint\plugin\core.py", line 78, in parse_plugin_metadata
    root = ast.parse(f.read(), filename=path)
  File "C:\Program Files\Python310\lib\ast.py", line 50, in parse
    return compile(source, filename, mode, flags,
  File "C:\Users\charl\Development\venv310\lib\site-packages\octoprint_slic3r\__init__.py", line 296
    if help_text_all[0].find(b'trace') >= 0
                                           ^
SyntaxError: expected ':'
2022-08-25 17:11:23,499 - octoprint.plugin.core - WARNING - Plugin OctoPrint-Slic3r (1.3.0) can't be compiled under Python 3.10.2 due to invalid syntax
```

solo2424, 2022-08-25:
Thank you for the reply! You have any idea how to resolve this

cp2004, 2022-08-25:
The issue should be resolved on the master branch (#71). A reinstall of the plugin should pick up the fixes.

eyal0, 2022-08-25:
> The issue should be resolved on the master branch (#71). A reinstall of the plugin should pick up the fixes.

I might need to make a release. I think that octoprint takes from the releases, not from `master`.

eyal0, 2022-08-29:
Fixed by #71

closed 2022-08-29 by eyal0