OctoPrint/OctoPrint-Slic3r #72: Ipad

cabbagemec, 2023-03-18:
Hi all. 
Can this be loaded onto a mac to then be loaded to Octoprint? In a similar way as on the Windows system in the instructions.

eyal0, 2023-03-20:
It might work!  Try it!

cabbagemec, 2023-03-25:
Ok after a day of trying, even going back to a complete reinstall twice of OctoPrint as well. I could get a pass on installation but the path to it could never be found. I am not a coding person having minimal knowledge.  I tried slic3r 1.2.9 and 1.1.7. The latter being the only one that would pass installation.

My son turned up with his laptop and in a few minutes installed the Prusa 2.4.1 appImage, (suppose it helps have a doctorate in web design)  as I use a PrusaSlicer and the path to this worked and would slice but would not print. Throwing up a missing layer issue. In conjunction ok the slicer plugin also worked with it but still not printing.

cabbagemec, 2023-03-26:
It would appear the error message was due to a plugin, but I am unsure which one as I removed 3, Still not printing though.